espressif/esp-mdf #201: [esp-mdf-master\examples\development_kit\light工程的ESPMesh连接配网问题]

jayjayhust, 2021-02-20:
## Environment

Development Kit: [ESP32-DevKitC]
Kit version (for WroverKit/PicoKit/DevKitC): [v4]
Core (if using chip or module): [ESP-WROOM-32D]
MDF version (git rev-parse --short HEAD to get the commit id.): v1.0
Development Env: [other]
Operating System: [Windows]
Power Supply: [USB]

## Problem Description

esp-mdf-master\examples\development_kit\light工程用ESPMesh(APP和小程序都试了)扫描不到设备（用手机本身的蓝牙设置菜单对附近蓝牙设备扫描，是可以看到设备的，我的是华为安卓机），所以无法完成配网，但是esp-mdf-master\examples\function_demo\mconfig工程就可以用ESPMesh扫描到设备然后完成配网。请问大家有碰到同样的问题吗？

peter81015, 2021-02-21:
我也遇到了，手机是OPPP reno5，也是手机自身的蓝牙扫描能发现light，但用微信版的小程序就不行。安卓版的APP更糟糕，经常卡在启动画面不动，进不了程序主界面

jayjayhust, 2021-02-21:
> 我也遇到了，手机是OPPP reno5，也是手机自身的蓝牙扫描能发现light，但用微信版的小程序就不行。安卓版的APP更糟糕，经常卡在启动画面不动，进不了程序主界面

安卓APP卡在启动界面一般是之前进去APP的时候，在“我的”标签页，点击了“云端”，后面再进APP，基本就会卡在启动界面了，需要卸载重装，没来得及看APP代码，不知道是什么原因导致的。

应该不是手机的问题，因为同样的手机，esp-mdf-master\examples\development_kit\light工程不正常，但是master\examples\function_demo\mconfig工程是可以用ESPMesh配网的。开始怀疑是两个工程的sdconfig文件对于硬件的配置不一致导致的，但是仔细逐行比对后，返现没有不一致的地方，尤其对于蓝牙的设置。今天在研读底层代码，mconfig_blufi.c和esp_gap_ble_api.c，准备查看device广播数据--->APP收到广播、发送scan数据--->device回复scan数据这个流程卡在哪里了，车补预测应该是“APP收到广播、发送scan数据--->device回复scan数据”这个步骤了，准备找到这部分通讯代码，打印出日志进行比对分析。

jayjayhust, 2021-02-23:
今天终于定位到问题了，就是esp-mdf-master\examples\development_kit\light工程不能menuconfig里面不能使能“new aliyun sdk IoT Platform”，否则ESPMesh就扫描不到设备。但是不使能“new aliyun sdk IoT Platform”，又不能编译aliyun_light_example.c（会报错找不到aliyun驱动头文件），请问要如何解决呢？或者说使能“new aliyun sdk IoT Platform”后里面的子菜单要做特殊配置？
[image]

jayjayhust, 2021-02-24:
还是esp-mdf-master\examples\development_kit\light工程，没用aliyun_light_example.c了，跑的light_example.c，这下ESPMesh小程序和APP都能扫描到模块了，不过设置的时候卡在了手机向模块发送WIFI配置信息，手机小程序报错如下图：
[image]
[image]

模块中控台的报错如下图：
[image]

请教下大神们怎么解决呢？

jayjayhust, 2021-02-24:
应该是IDF版本和MDF版本匹配的问题，又把工程重新下载了，目前是可以了：
https://github.com/espressif/esp-mdf/releases/tag/v1.0

closed 2021-02-24 by jayjayhust

whwxxstsyd, 2021-05-11:
遇到同样的问题，谢谢楼主的解答，可以请教下是怎么定位到“new aliyun sdk IoT Platform”这个配置下的