espressif/esp-mdf #202: master branch: light 例子，新root节点不能获取ip

rqbh, 2021-03-03:
最新的master branch MDF, 两个节点，把根节点掉电后，另一个节点转换为新的根节点，但不能获取ip
日志：
I (1334171) [mesh_utils, 157]: System information, channel: [12/2], layer: 2, self mac: 84:cc:a8:61:03:1c, parent bssid: 3c:61:05:0c:e1:7d, parent rssi: -67, node num: 2, free heap: 140596
I (1344171) [mesh_utils, 157]: System information, channel: [12/2], layer: 2, self mac: 84:cc:a8:61:03:1c, parent bssid: 3c:61:05:0c:e1:7d, parent rssi: -67, node num: 2, free heap: 140472
I (1351213) wifi:bcn_timout,ap_probe_send_start
I (1353717) wifi:ap_probe_send over, resett wifi status to disassoc
I (1353719) wifi:state: run -> init (c800)
I (1353720) wifi:pm stop, total sleep time: 0 us / 246382354 us

I (1353732) wifi:new:<12,0>, old:<12,2>, ap:<12,2>, sta:<12,0>, prof:12
I (1353735) [mwifi, 187]: Parent is disconnected, reason: 200
I (1353737) [light_example, 419]: event_loop_cb, event: 0x8
I (1353737) mesh: [wifi]disconnected reason:200(beacon timeout), continuous:1/max:12, non-root, vote(,stopped)<><>
I (1353749) [light_example, 503]: Parent is disconnected on station interface
I (1353761) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (1353775) wifi:new:<12,2>, old:<12,0>, ap:<12,2>, sta:<12,0>, prof:12
I (1353777) wifi:state: init -> auth (b0)
I (1354171) [mesh_utils, 157]: System information, channel: [12/2], layer: -1, self mac: 84:cc:a8:61:03:1c, parent bssid: 3c:61:05:0c:e1:7d, parent rssi: -67, node num: 2, free heap: 140960
I (1354979) wifi:state: auth -> init (200)
I (1354981) wifi:new:<12,0>, old:<12,2>, ap:<12,2>, sta:<12,0>, prof:12
I (1354984) [mwifi, 187]: Parent is disconnected, reason: 2
I (1354985) [light_example, 419]: event_loop_cb, event: 0x8
I (1354985) mesh: [wifi]disconnected reason:2(auth expire), continuous:2/max:12, non-root, vote(,stopped)<><>
I (1354997) [light_example, 503]: Parent is disconnected on station interface
I (1355010) wifi:new:<12,2>, old:<12,0>, ap:<12,2>, sta:<12,0>, prof:12
I (1355022) wifi:state: init -> auth (b0)
I (1356224) wifi:state: auth -> init (200)
I (1356226) wifi:new:<12,0>, old:<12,2>, ap:<12,2>, sta:<12,0>, prof:12
I (1356229) [mwifi, 187]: Parent is disconnected, reason: 2
I (1356230) mesh: [wifi]disconnected reason:2(auth expire), continuous:3/max:12, non-root, vote(,stopped)<><>
I (1356241) [light_example, 419]: event_loop_cb, event: 0x8
I (1356243) wifi:new:<12,2>, old:<12,0>, ap:<12,2>, sta:<12,0>, prof:12
I (1356255) wifi:state: init -> auth (b0)
I (1356256) [light_example, 503]: Parent is disconnected on station interface
I (1357467) wifi:state: auth -> init (200)
I (1357469) wifi:new:<12,0>, old:<12,2>, ap:<12,2>, sta:<12,0>, prof:12
I (1357472) [mwifi, 187]: Parent is disconnected, reason: 2
I (1357473) mesh: [wifi]disconnected reason:2(auth expire), continuous:4/max:12, non-root, vote(,stopped)<><>
I (1357484) [light_example, 419]: event_loop_cb, event: 0x8
I (1357486) wifi:new:<12,2>, old:<12,0>, ap:<12,2>, sta:<12,0>, prof:12
I (1357498) wifi:state: init -> auth (b0)
I (1357500) [light_example, 503]: Parent is disconnected on station interface
I (1358700) wifi:state: auth -> init (200)
I (1358702) wifi:new:<12,0>, old:<12,2>, ap:<12,2>, sta:<12,0>, prof:12
I (1358705) [mwifi, 187]: Parent is disconnected, reason: 2
I (1358706) mesh: [wifi]disconnected reason:2(auth expire), continuous:5/max:12, non-root, vote(,stopped)<><>
I (1358717) [light_example, 419]: event_loop_cb, event: 0x8
I (1358719) wifi:new:<12,2>, old:<12,0>, ap:<12,2>, sta:<12,0>, prof:12
I (1358731) wifi:state: init -> auth (b0)
I (1358733) [light_example, 503]: Parent is disconnected on station interface
I (1359933) wifi:state: auth -> init (200)
I (1359934) wifi:new:<12,0>, old:<12,2>, ap:<12,2>, sta:<12,0>, prof:12
I (1359937) [mwifi, 187]: Parent is disconnected, reason: 2
I (1359949) [light_example, 419]: event_loop_cb, event: 0x8
I (1359951) [light_example, 503]: Parent is disconnected on station interface
I (1359953) [mwifi, 265]: Network state: root_less
I (1359963) [light_example, 419]: event_loop_cb, event: 0xb
I (1359953) mesh: 5162[healing]looking for a new parent, [L:2]try layer:1[revote][scan]
I (1359975) [light_example, 419]: event_loop_cb, event: 0x14
I (1360578) mesh: [SCAN][ch:12]AP:1, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:2,i:1][00:5a:39:19:51:8e]router found<>
I (1360581) mesh: 1315[SCAN]init rc[84:cc:a8:61:03:1d,-86], mine:0, voter:0
I (1360592) mesh: 1353, vote myself, router rssi:-86 > voted rc_rssi:-120
I (1360604) mesh: [SCAN:1/10]rc[128][84:cc:a8:61:03:1d,-86], self[84:cc:a8:61:03:1c,-86,reason:2,votes:1,idle][mine:1,voter:1(1.00)percent:0.90][128,1,84:cc:a8:61:03:1d]

I (1361219) mesh: [SCAN][ch:12]AP:1, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:2,i:1][00:5a:39:19:51:8e]router found<>
I (1361222) mesh: 1315[SCAN]init rc[84:cc:a8:61:03:1d,-84], mine:0, voter:0
I (1361234) mesh: [SCAN:2/10]rc[128][84:cc:a8:61:03:1d,-84], self[84:cc:a8:61:03:1c,-84,reason:2,votes:1,idle][mine:1,voter:1(1.00)percent:0.90][128,1,84:cc:a8:61:03:1d]

I (1361849) mesh: [SCAN][ch:12]AP:2, other(ID:0, RD:0), MAP:0, idle:0, candidate:1, root:0, topMAP:0[c:2,i:1][00:5a:39:19:51:8e]router found<>
I (1361852) mesh: 1315[SCAN]init rc[84:cc:a8:61:03:1d,-79], mine:0, voter:0
[... 189 lines cut ...]
I (2265695) mesh: [wifi]disconnected reason:8(assoc leave), continuous:0/max:12, root, vote(,stopped)<><>
I (2265696) [light_example, 503]: Parent is disconnected on station interface
I (2266310) wifi:new:<12,2>, old:<12,0>, ap:<12,2>, sta:<12,0>, prof:12
I (2266313) wifi:state: init -> auth (b0)
I (2266318) wifi:state: auth -> assoc (0)
I (2266336) wifi:state: assoc -> run (10)
I (2266710) wifi:connected with FAST_123, aid = 3, channel 12, BW20, bssid = 00:5a:39:19:51:8e
I (2266712) wifi:security: WPA2-PSK, phy: bgn, rssi: -76
I (2266725) wifi:pm start, type: 0

I (2266730) [mwifi, 174]: Parent is connected
I (2266731) [light_example, 419]: event_loop_cb, event: 0x7
I (2266732) [mwifi, 257]: State represents: 0
I (2266732) [light_example, 428]: Parent is connected on station interface
I (2266732) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (2266744) [light_example, 419]: event_loop_cb, event: 0xb
I (2266756) [light_example, 419]: event_loop_cb, event: 0xe
I (2266756) mesh: <flush>root
I (2266768) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(1, parent:3c:61:05:0c:e1:7d)
I (2266779) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (2266811) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2274171) [mesh_utils, 157]: System information, channel: [12/2], layer: 1, self mac: 84:cc:a8:61:03:1c, parent bssid: 00:5a:39:19:51:8e, parent rssi: -76, node num: 1, free heap: 140804
I (2284171) [mesh_utils, 157]: System information, channel: [12/2], layer: 1, self mac: 84:cc:a8:61:03:1c, parent bssid: 00:5a:39:19:51:8e, parent rssi: -76, node num: 1, free heap: 140804
I (2294171) [mesh_utils, 157]: System information, channel: [12/2], layer: 1, self mac: 84:cc:a8:61:03:1c, parent bssid: 00:5a:39:19:51:8e, parent rssi: -76, node num: 1, free heap: 140804
I (2304171) [mesh_utils, 157]: System information, channel: [12/2], layer: 1, self mac: 84:cc:a8:61:03:1c, parent bssid: 00:5a:39:19:51:8e, parent rssi: -82, node num: 1, free heap: 140804

rqbh, 2021-03-03:
MDF的release/v1.0 是没有这个问题的

SgrrZhf, 2021-03-24:
我测试了是可以的，你能换一个路由器测试下吗？