espressif/esp-mdf #205: mwifi_read does not support newline code

tatsutaigu, 2021-03-09:
## Environment

- Development Kit:      ESP32-WROVER-B
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 1c76187
- Operating System:		WSL (Windows Subsystem on Linux)
- Power Supply:         USB

## Problem Description
When mwifi_read gets newline code (CR, LF), mwifi_read replaced \r\n.
mwifi_write can write newline code(CR, LF).
TCP server could receive newline code(CR, LF).

I would like to use newline code and binary data because I would like to use UART to control via ESP_MESH.

### Expected Behavior

### Actual Behavior

### Steps to repropduce
1. to use below sample.
[https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router)

2. TCP server sends newline code(CR, LF)

## Debug Logs
here is log on monitor.
```
D (2957185) [router_example, 116]: TCP read, 58, size: 1456, data: {"dest_addr": "30:ae:a4:ce:59:70","data":"

"}

D (2957187) [router_example, 233]: Node receive: 30:ae:a4:ce:59:70, size: 10, data: "\r\n\r\n"
```

SgrrZhf, 2021-03-24:
JSON doesn't allow real line-breaks.  If you want to send binary data, you can base64 encode it.

tatsutaigu, 2021-03-24:
Thank you for your advice.
I will use base64 enc/dec.

closed 2021-03-24 by tatsutaigu