espressif/esp-mdf #207: cmake failed with exit code 1

Lss615, 2021-03-21:
环境搭建后，试编译hello world程序时，输入idf.py build后，出现如下界面，
[image]
[image]
求问该如何解决

SgrrZhf, 2021-03-25:
建议你使用 powershell 重新搭建环境试一下

Lss615, 2021-03-28:
您好，
1.请问，在power shell下无法识别python程序怎么办？我们根据网上的提示尝试将python.exe所在的路径加到系统环境变量Path中了，但是还是不行。
2.power shell下对应cmd set MDF_PATH=""的操作是$Env:path=$Env:Path+";"吗

EspHuifeng, 2021-03-29:
您也可以输入完整的 python 路径来运行 python。powershell 的使用方法，建议你去阅读官方的文档。

Lss615, 2021-03-29:
谢谢帮助，我们还遇到了一个esp网络套件那个app的问题

就是我们在连接灯和开关的时候，每次都是到60%显示错误（app显示wifi密码不正确，微信小程序显示的是蓝牙连接中断)，请问是什么原因，十分感谢。

EspHuifeng, 2021-03-29:
能否提供设备测的 log 看看

Lss615, 2021-03-30:
啊，我们这个是用app连接的meshkit-light,想看一下效果，已经解决了。
就是我们之前都是在安卓系统手机的app上连接尝试的，今天用iPad试了一下，发现成功了。这个app是不是更适配ios系统？
十分感谢您的帮助。
respect。