espressif/esp-mdf #209: what example code to use?

sebasdt, 2021-03-26:
Hi!

In the past week, I have been trying to learn how the mesh and some examples work.
one thing isn't very clear for me what example code I should use to get a basic mesh working with the ESP-MESH app functionality.

is it needed to combine some provided examples?

thanks in advance!

EspHuifeng, 2021-03-29:
Please reference [Mesh-Light example](https://github.com/espressif/esp-mdf/tree/master/examples/development_kit/light#esp32-meshkit-light)

sebasdt, 2021-03-31:
thanks got it working.
i also discoverd location is needed, could that be added to the readME? please.

sebasdt, 2021-03-31:
https://skribbl.io/?S2EbUmHUym0G

sebasdt, 2021-03-31:
thanks!

could you also add in readme to enable location. this wasn't clear to me.