espressif/esp-mdf #212: Mupgrade & Flash Encryption Issue

MrCurio, 2021-04-09:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): master
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
Hello everyone, 
I am developing an application where **OTA** is a requirement. Since all nodes are not in the range of the router, i decided to use **MUPGRADE**. So far i made all my tests and every ran as expected. However, we are getting ready for production and i've included both Secure Boot and Flash Encryption. The problem comes when i try to use **Mupgrade with Flash Encryption**. When it is it enabled, writing to the flash fails at the point where there's a packet loss between neighbour nodes. 

I was able to track the issue until here:

```
mupgrade_node.c

    /**< Write firmware data to the update partition */
    ret = esp_partition_write(g_upgrade_config->partition, packet->seq * MUPGRADE_PACKET_MAX_SIZE,
                              packet->data, packet->size);
```

```
flash_ops.c

esp_err_t IRAM_ATTR spi_flash_write_encrypted(size_t dest_addr, const void *src, size_t size)
{
    esp_err_t err = ESP_OK;
    CHECK_WRITE_ADDRESS(dest_addr, size);
    if ((dest_addr % 16) != 0) {
        return ESP_ERR_INVALID_ARG;
    }
    if ((size % 16) != 0) { -------------> When there is a packet loss the size might not be 1024
        return ESP_ERR_INVALID_SIZE;
    }

    COUNTER_START();
    esp_rom_spiflash_result_t rc = spi_flash_unlock();
    err = spi_flash_translate_rc(rc);
    if (err != ESP_OK) {
        goto fail;
    }

```

In my script everything runs smoothly until there is a packet with size different from 1024 (i assume due to packet loss). After that, the function ``` esp_partition_write(...) ``` starts to repeatedly  return the error: ``` ESP_ERR_INVALID_SIZE ``` and everything stops working.

### Expected Behavior
Mupgrade working with flash encryption enabled

### Actual Behavior
When Flash encryption is enabled the program is not working

### Steps to repropduce

1. Flash mupgrade example with flash encryption enabled 

## Debug Logs

```
I (187107) mupgrade_node: Packet size: 1012  Offset: 1702912    Partition size: 3145728
W (187117) [mupgrade_node, 194]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> esp_partition_write ESP_ERR_INVALID_SIZE
W (187127) [mupgrade_node, 280]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
W (187137) [Mupgrade, 76]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
D (189717) [mupgrade_node, 128]: Response mupgrade status, written_size: 1702912, response_size: 557
I (189797) mupgrade_node: Packet size: 1012  Offset: 1702912    Partition size: 3145728
W (189797) [mupgrade_node, 194]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> esp_partition_write ESP_ERR_INVALID_SIZE
W (189807) [mupgrade_node, 280]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
W (189817) [Mupgrade, 76]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
D (192797) [mupgrade_node, 128]: Response mupgrade status, written_size: 1702912, response_size: 557
I (192887) mupgrade_node: Packet size: 1012  Offset: 1702912    Partition size: 3145728
W (192887) [mupgrade_node, 194]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> esp_partition_write ESP_ERR_INVALID_SIZE
W (192897) [mupgrade_node, 280]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
W (192897) [Mupgrade, 76]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
D (195887) [mupgrade_node, 128]: Response mupgrade status, written_size: 1702912, response_size: 557
I (195967) mupgrade_node: Packet size: 1012  Offset: 1702912    Partition size: 3145728
W (195967) [mupgrade_node, 194]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> esp_partition_write ESP_ERR_INVALID_SIZE
W (195977) [mupgrade_node, 280]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
W (195987) [Mupgrade, 76]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
D (198977) [mupgrade_node, 128]: Response mupgrade status, written_size: 1702912, response_size: 557
I (199067) mupgrade_node: Packet size: 1012  Offset: 1702912    Partition size: 3145728
W (199067) [mupgrade_node, 194]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> esp_partition_write ESP_ERR_INVALID_SIZE
W (199077) [mupgrade_node, 280]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
W (199077) [Mupgrade, 76]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
D (202087) [mupgrade_node, 128]: Response mupgrade status, written_size: 1702912, response_size: 557
I (202167) mupgrade_node: Packet size: 1012  Offset: 1702912    Partition size: 3145728
W (202167) [mupgrade_node, 194]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> esp_partition_write ESP_ERR_INVALID_SIZE
W (202177) [mupgrade_node, 280]: <MDF_ERR_MUPGRADE_FIRMWARE_DOWNLOAD> mupgrade_handle
```

InfiniteYuan, 2021-05-25:
You should use esp_ota_* API instead. You can manually modify the relevant code according to this example(esp-idf/examples/system/ota/native_ota_example/main/native_ota_example.c). 
The official will complete the update later. 

Thank you for your feedback.

MrCurio, 2021-05-25:
Thanks @InfiniteYuan !
So far i will continue with the workaround cause it implies less coding :), but if i have further issues, i will try as you mention.

I proceed to close this issue

closed 2021-05-25 by MrCurio