espressif/esp-mdf #217: esp-mdf-master\examples\development_kit\light工程aliyun_light蓝牙配网使用mesh-app找不到信号

whwxxstsyd, 2021-05-12:
esp-mdf-master\examples\development_kit\light工程aliyun_light例程，使用mesh-app进行配网时，找不到蓝牙信号，导致无法配网，但是用普通的light例程又是可以的

两个例程蓝牙信号都是有的，在手机设备蓝牙内都可以发现

[image]

之前好像也有人提过类似的问题，是因为开启了“new aliyun sdk IoT Platform”这个配置选项，但是如果不开启就没办法使用aliyun_light的例程，求解

esp-lqq, 2021-05-12:
您好：
公版的 mesh-app 不支持对 mesh aliyun 设备配网，因为 APP 需要单独编译以支持aliyun。
IOS： https://github.com/EspressifApp/EspMeshForiOS
安卓：https://github.com/EspressifApp/EspMeshForAndroid

whwxxstsyd, 2021-05-12:
您好，我比对了mesh aliyun light和普通mesh light的代码，配网部分的代码并没有发现什么差异

公版的 mesh-app 为什么连mesh aliyun的蓝牙信号都发现不了，很奇怪，是在哪里做了区分吗

esp-lqq, 2021-05-12:
是的，APP 做了区分，设备蓝牙包存在不同的标志。
在 aliyun sdk kconfig 中 会配置 BLUFI_BROADCAST_OUI 参数为 MAY，APP 在收到这个信息时会进入 aliyun 模式。默认 公版 APP 不识别该类型。

whwxxstsyd, 2021-05-12:
感谢您的回答，找到问题的答案了，以下做个记录

在aliyun sdk kconfig 中找到BLUFI_BROADCAST_OUI配置参数，默认为 MAY
 
[image]

在EspMeshForAndroid中也找到了相关的参数定义

[image]

在mdf的api文档中也可以找到OUI具体的定义，公版 APP 默认的过滤字段和aliyun mesh的过滤字段不一样，所以才导致找不到设备

[image]

closed 2021-05-12 by whwxxstsyd