espressif/esp-mdf #219: How to improve throughput of console_test?

bkkim007, 2021-05-18:
----------------------------- Delete below -----------------------------

How do I improve throughput result on console_test example code?
My test result is using two board : Max throughput is 0.75Mbps.

is there any way to improve max throughput on console_test example?

Thank you in advance
BK

zhanzhaocheng, 2021-05-20:
The communication speed of Wi-Fi Mesh will be interfered by the surrounding network environment. You can test or modify the channel of the Wi-Fi Mesh network in a clean network environment.

We currently test that 512 devices can work stably on a network. Our customers generally use less than 256 devices in actual use.

bkkim007, 2021-05-20:
Hi zhanzhaocheng,

thank you for your comment. I know the communication speed is very related to surrounding network environment. In my case, i tested on office and distance is very short. it is around 30cm between device and device

Under 1:1 mesh network environment, what is the Max communication speed and how can I improve this by software modification?

I assumed that any surrounding network environment are not impact to max communication speed about my question.

thank you
BK