espressif/esp-mdf #220: mdf的light历程，用app进行蓝牙配网，配网成功，没有显示

whwxxstsyd, 2021-05-19:
设备端用ESP-MDF-V1.0，使用“examples/development_kit/light”例程，APP端使用 “mesh-1.2.3-51.apk” 进行蓝牙配网，可以配网成功，但是APP上没有显示，下拉刷新也看不到设备

设备端Log：

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6812
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.2-14-g6c9c253-dirty 2nd stage bootloader
I (39) boot: compile time 00:54:58
I (39) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (98) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (105) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (113) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (121) boot: End of partition table
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x51570 (333168) map
I (258) esp_image: segment 1: paddr=0x00061598 vaddr=0x3ffbdb60 size=0x03ff0 ( 16368) load
I (265) esp_image: segment 2: paddr=0x00065590 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/book/esp/esp-mdf-v1.0/esp-idf/components/freertos/xtensa_vectors.S:1779

I (265) esp_image: segment 3: paddr=0x00065998 vaddr=0x40080400 size=0x0a678 ( 42616) load
I (291) esp_image: segment 4: paddr=0x00070018 vaddr=0x400d0018 size=0x121618 (1185304) map
0x400d0018: _flash_cache_start at ??:?

I (708) esp_image: segment 5: paddr=0x00191638 vaddr=0x4008aa78 size=0x11b64 ( 72548) load
0x4008aa78: r_lld_evt_restart at ??:?

I (754) boot: Loaded app from partition at offset 0x10000
I (755) boot: Disabling RNG early entropy source...
I (755) cpu_start: Pro cpu up.
I (759) cpu_start: Application information:
I (763) cpu_start: Project name:     light
I (768) cpu_start: App version:      v1.0-dirty
I (773) cpu_start: Compile time:     May 19 2021 00:55:06
I (780) cpu_start: ELF file SHA256:  18961203d172956c...
I (786) cpu_start: ESP-IDF:          v3.3.2-14-g6c9c253-dirty
I (792) cpu_start: Single core mode
D (796) memory_layout: Checking 9 reserved memory ranges:
D (802) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (808) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (814) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (821) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (827) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (834) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffcab50
D (840) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (847) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (853) memory_layout: Reserved memory range 0x40080000 - 0x4009c5dc
0x40080000: _WindowOverflow4 at /home/book/esp/esp-mdf-v1.0/esp-idf/components/freertos/xtensa_vectors.S:1779

D (859) memory_layout: Building list of available memory regions:
D (866) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
[... 294 lines cut ...]

{"status_code":-1,"status_msg":"Destination address error, No device response"}
E (64927) httpd_txrx: httpd_sock_err: error calling getsockopt : 9
D (65124) [light_example, 124]: Root send, addrs_num: 1, addrs_list: 00:00:00:00:00:00, size: 29, data: {"request":"get_device_info"}
E (65127) [mwifi, 1106]: <MDF_ERR_INVALID_ARG> !(!MWIFI_ADDR_IS_EMPTY(addrs_list))
W (65138) [light_example, 131]: <MDF_ERR_INVALID_ARG> mwifi_root_write
I (67950) [mesh_utils, 149]: System information, channel: [11/2], layer: 1, self mac: 08:3a:f2:ac:66:58, parent bssid: 9c:a6:15:0d:b7:6c, parent rssi: -56, node num: 1, free heap: 98484
W (73600) [mlink_httpd, 182]: Mlink httpd response timeout, sockfd: 61, data: HTTP/1.1 400 Bad Request
Content-Type: application/json
Content-Length: 79

{"status_code":-1,"status_msg":"Destination address error, No device response"}
E (73617) httpd_txrx: httpd_sock_err: error calling getsockopt : 9
D (73726) [light_example, 124]: Root send, addrs_num: 1, addrs_list: 00:00:00:00:00:00, size: 29, data: {"request":"get_device_info"}
E (73728) [mwifi, 1106]: <MDF_ERR_INVALID_ARG> !(!MWIFI_ADDR_IS_EMPTY(addrs_list))
W (73740) [light_example, 131]: <MDF_ERR_INVALID_ARG> mwifi_root_write
I (77950) [mesh_utils, 149]: System information, channel: [11/2], layer: 1, self mac: 08:3a:f2:ac:66:58, parent bssid: 9c:a6:15:0d:b7:6c, parent rssi: -56, node num: 1, free heap: 98192
W (80126) [mlink_httpd, 182]: Mlink httpd response timeout, sockfd: 62, data: HTTP/1.1 400 Bad Request
Content-Type: application/json
Content-Length: 79

{"status_code":-1,"status_msg":"Destination address error, No device response"}
E (80142) httpd_txrx: httpd_sock_err: error calling getsockopt : 9

book@ubuntu:~/esp/esp-mdf-v1.0/examples/development_kit/light$

zhanzhaocheng, 2021-05-20:
这个问题在 master 分支上已经修复了，我们会尽快 backport 到 v1.0

whwxxstsyd, 2021-05-21:
那请问一下，如果我想基于一个版本的mdf做产品开发，那用master分支还是v1.0

zhanzhaocheng, 2021-05-21:
属于刚启动的项目推荐基于 master 开发，
v1.0 只支持 esp-idf v3.3，而且 master 中的 esp-idf 之后会更新到 v4.3，可支持 ESP32-C3、ESP32-S3

whwxxstsyd, 2021-05-21:
好的，谢谢

closed 2021-05-21 by whwxxstsyd