espressif/esp-mdf #224: Cannot send to a group unless data type is set to BROADCAST, examples do not have this.

sagar448, 2021-06-25:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP32-WROOM-32E]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    release 1.0 branch
- Development Env:		[IDF]
- Operating System:		[Windows]
- Power Supply:         [USB|external 5V|]

## Problem Description
Currently the root is setup on ethernet, we're using the fixed root. There are currently 2 other devices in the mesh.
Following the root on ethernet example, tried to send to a group with a specified group ID. In the examples the data_type is set to {0x0} and then the group is set to true. IF we do this, it does not work. But if we set the .communicate to BROADCAST, it works but 50% of the packets are dropped. So it receives it 50% of the time, and the rest of the time the nodes do not receive this. 

### Expected Behavior
Message should be received by the nodes 100% of the times.

### Actual Behavior
The packets are dropped and only 50% of the group messages are received. E.g. if we send 200 messages, one node will receive about 90 messages, the other node will receive around a 100 messages.

### Code to reproduce this issue

This goes in the main file.

```cpp
const uint8_t group_id_list[6] = {0x01, 0x00, 0x5e, 0xae, 0xae, 0xae};
    if (esp_mesh_set_group_id((mesh_addr_t *)group_id_list, 1) == ESP_OK) ESP_LOGI(TAG, "Group new loaded");

```

Then we are using this in the root 

```
const uint8_t group_id_list[6] = {0x01, 0x00, 0x5e, 0xae, 0xae, 0xae};
    mdf_err_t ret = mwifi_write(group_id_list, &data_type, (char *)data, size, true);
```

Everything is followed with the example, except the ethernet settings, for the mode we're using APSTA instead of STA as in the example.

Any help will be greatly appreciated!

@EspHuifeng

sagar448, 2021-06-29:
Hey can anyone help with this please?

zhanzhaocheng, 2021-07-06:
I'm very sorry, I'm just replying

This is because non-broadcast multicast needs to be forwarded through the bottom layer of `mwifi_read`, you can comment the following lines:https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mwifi/root_on_ethernet/main/ethernet_example.c#L197-L200

```c

+++ b/examples/function_demo/mwifi/root_on_ethernet/main/ethernet_example.c
@@ -194,10 +194,10 @@ static void node_read_task(void *arg)
     MDF_LOGI("Note read task is running");
 
     for (;;) {
-        if (!g_root_got_ip) {
-            vTaskDelay(500 / portTICK_RATE_MS);
-            continue;
-        }
+        // if (!g_root_got_ip) {
+        //     vTaskDelay(500 / portTICK_RATE_MS);
+        //     continue;
+        // }
 ```

sagar448, 2021-07-15:
No problem.
OK will try this and let you know, thank you @zhanzhaocheng 
Is it possible for me to message you directly somewhere? That would be really helpful.

sagar448, 2021-07-15:
Actually just checked the code, I do not have those lines

I have this instead

`if (!mwifi_is_connected() && !(mwifi_is_started() && esp_mesh_is_root())) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }`

zhanzhaocheng, 2021-07-16:
If you want to use `non-broadcast multicast`, you must keep mwifi_read of the root node running. It is recommended that you change it to:

```c
if (!mwifi_is_started()) {
       vTaskDelay(500 / portTICK_RATE_MS);
       continue;
}
```

If you still can’t solve your problem, you can provide your code and I will debug on your basis

sagar448, 2021-07-21:
Hello @zhanzhaocheng Apologies for the late response.
That seems to have worked along with the communicate set to broadcast and the group set to true.
Unfortunately theres heavy packet drop, is there a solution to that? We would like to minimize packet drop.
Another problem I noticed is, a group with more than 5 nodes, the packet drop is even more significant!

Almost 1 out of every 5 packet is dropped. Is there anything we can do to solve this, please let me know.

Thanks

sagar448, 2021-07-21:
As you mentioned, you said to use non-broadcast multicast, but it only works when communicate is set to broadcast. I can send you the code, but its quite a large codebase, it would be good if we can get on maybe a screenshare, that would help a lot. Please and thanks.

zhanzhaocheng, 2021-07-27:
@sagar448 Sorry that remote sharing is not very convenient, you can send the reproducible code snippets directly to my mailbox (zhanzhaozheng@espressif.com),
Or send me your needs, and I will provide you with sample projects based on your needs

sagar448, 2021-08-05:
Thank you @zhanzhaocheng I will email you. I'm closing this thread then.

closed 2021-08-05 by sagar448