espressif/esp-mdf #227: 无法导入esp-idf/components内的库

mynameisjyy, 2021-07-16:
1，下载的是最新的esp-mdf
2，在示例程序get-started上，加入idf示例程序esp-http-client部分内容,
3，报如下故障：../main/get_started.c:48:10: fatal error: esp_tls.h: No such file or directory
 #include "esp_tls.h"
帮忙查看下如何解决，谢谢！
[get_started.zip](https://github.com/espressif/esp-mdf/files/6829054/get_started.zip)

mynameisjyy, 2021-07-16:
只需要在get_started.c里  引入头文件  #include "esp_tls.h"  编译，就会出现报警。新人，帮忙看下如何解决，谢谢！

zhanzhaocheng, 2021-07-16:
在 `get_started` 工程的 CMakeLists.txt 加入 `esp_tls`

```cmake
idf_component_register(SRCS "./get_started.c"
                    INCLUDE_DIRS ""
                    REQUIRES mcommon mconfig mwifi mlink esp_tls
)
```

可以参考：https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router

mynameisjyy, 2021-07-17:
HI, zhan:
   你好，根据你的回复，我进行了测试，还是报警：
[image]
[image]
帮忙分析下原因。

zhanzhaocheng, 2021-07-20:
把 CMakeLists  中 esp_tls 改成 esp-tls 就可以了，已经基于你的代码验证了

mynameisjyy, 2021-07-21:
HI, zhan，已经测试，能正常编译，感谢！
再请教个事，我需要向一个远程后台一直定期发送数据（用http post)，可以参考哪个示例？

zhanzhaocheng, 2021-07-22:
http post: https://github.com/espressif/esp-idf/tree/e9cf9e2978d0187b0f83a9978391217e39946517/examples/protocols/esp_http_client

mynameisjyy, 2021-08-02:
谢谢，有问题再请教。

closed 2021-08-02 by mynameisjyy