espressif/esp-mdf #232: 使用定时器问题

waqq1314, 2021-07-31:
设备是mesh组网，其他功能都已经实现，当前调试的功能是解码433无线数据，需要定时器100us。
先说现象，没用定时器还好，只要使用就会崩溃，如果时间长还好，偶发，时间短100%出现。下面是代码

`static void timer_example_evt_task(void *arg)
{
	uint8_t level=0;
    while (1) {
        timer_event_t evt;
        xQueueReceive(timer_queue, &evt, portMAX_DELAY);
		gpio_set_level(21,level);
		level = !level;
		//ESP_LOGI(TAG, "t act");
    }
}

void IRAM_ATTR timer_group0_isr(void *para)
{
	timer_spinlock_take(TIMER_GROUP_0);
    int timer_idx = (int) para;
	timer_event_t evt;
    /* Retrieve the interrupt status and the counter value
       from the timer that reported the interrupt */
    //uint32_t timer_intr = timer_group_get_intr_status_in_isr(TIMER_GROUP_0);
    //uint64_t timer_counter_value = timer_group_get_counter_value_in_isr(TIMER_GROUP_0, timer_idx);
	timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, timer_idx);
	timer_group_enable_alarm_in_isr(TIMER_GROUP_0, timer_idx);
	xQueueSendFromISR(timer_queue, &evt, NULL);
	//uint8_t pin = gpio_get_level(21);
	//LittleWirelessDecode(pin);
	//ESP_LOGI(TAG, "t act");
    /* After the alarm has been triggered
      we need enable it again, so it is triggered the next time */
    timer_spinlock_give(TIMER_GROUP_0);
}

void TimerInit(timer_group_t tgroup,timer_idx_t timer)
{
	gpio_config_t io_conf;
	
	//interrupt of rising edge
	io_conf.intr_type = GPIO_INTR_DISABLE;
	//bit mask of the pins
	io_conf.pin_bit_mask = 1<<21;
	//set as input mode
	io_conf.mode = GPIO_MODE_OUTPUT;
	//enable pull-up mode
	io_conf.pull_up_en = 0;
	gpio_config(&io_conf);
	
	/*hw_timer_init(hw_timer_callback, NULL);
	hw_timer_alarm_us(100, true);
	timer_set_alarm_value*/

	/* Select and initialize basic parameters of the timer */
    timer_config_t config = {
        .divider = 16,
        .counter_dir = TIMER_COUNT_UP,
        .counter_en = TIMER_PAUSE,
        .alarm_en = TIMER_ALARM_EN,
		.intr_type = TIMER_INTR_LEVEL,
        .auto_reload = true,
    }; // default clock source is APB
    esp_err_t ret = timer_init(tgroup, timer, &config);
	ESP_LOGI(TAG, "res %d",ret);
    /* Timer's counter will initially start from value below.
       Also, if auto_reload is set, this value will be automatically reload on alarm */
    ret = timer_set_counter_value(tgroup, timer, 0x00000000ULL);
	ESP_LOGI(TAG, "res %d",ret);
    /* Configure the alarm value and the interrupt on alarm. */
    ret = timer_set_alarm_value(tgroup, timer, (TIMER_BASE_CLK / 16)/10000);
	ESP_LOGI(TAG, "res %d",ret);
    ret = timer_enable_intr(tgroup, timer);
	ESP_LOGI(TAG, "res %d",ret);
    ret = timer_isr_register(tgroup, timer, timer_group0_isr,
                       (void *) timer, ESP_INTR_FLAG_IRAM, NULL);
	ESP_LOGI(TAG, "res %d",ret);
    ret = timer_start(tgroup, timer);
	ESP_LOGI(TAG, "res %d",ret);
}`

这里用io翻转测试速度，在1ms时还只是偶发崩溃，并且波形也正常，但100us后必然崩溃，并且波形也不正常了。

之前用8266时用的hw_timer,完全没问题。
这里是有什么兼容问题吗？

esp-lqq, 2021-07-31:
你好，433 协议建议您使用 RMT 外设做编解码，ESP32S 系列系列具备该外设 。
另外其他芯片可以通过 IO 中断，读取 系统时间软件解码，解码不需要使用硬件定时器。

waqq1314, 2021-08-02:
也就是说定时器无解咯！
因为之前一直用的定时器，就打算直接代码移植，如果这样的话我就得重新搞了

closed 2021-12-23 by waqq1314