espressif/esp-mdf #235: Part of packet is lost epect_seq : 0 , recv_seq: 1

ehabulhaq, 2021-08-15:
## Environment

- Development Kit:     ESP32-LyraTD-MSC
- Core (if using chip or module): |ESP32-Wrover|
- MDF version 
- Development Env:		VSCode 
- Operating System:		[Windows|
- Power Supply:         [USB|

## Problem Description

I am designing a wireless mesh audio system.  There is a master root node that sends data to child nodes via multicast broadcast. At one time the master node sends (7 * 1024) bytes.  The child node at layer 2 forwards the packet to layer 3 node. 

I only have three nodes for now for testing.

At master there is a task that runs continuously to send data to children nodes all the time with same size.

At nodes there is a task that receives data from parent node and send it to I2S buffer.

The problem occurs after sometime when layer 3 child node falls into an infinte loop where it prints the Warning statement:

**W (48656) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 2**

I am attaching a debug log of layer 2 node and layer 3 node:

you will notice the layer 3 node starts receving the same packet twice. its expecting 1 and the layer 2 sends it pckt 0 again and it fails to receive

### Steps to repropduce

1.  create a task on root node to write (7 * 1024 ) bytes to child nodes via multicast
2. force only single child node on station interface

3. create a task on child nodes to receive at layer 2 and layer 3 
4. wait for layer 3 node to fall into infinite loop

### Code to reproduce this issue

```cpp
//Root node task where it writes
static void meshFeed(void *pvParameters)
{
    mdf_err_t ret = MDF_OK;
    char *data = (char*) MDF_MALLOC(MWIFI_PAYLOAD_LEN);
    while (true)
    {
        char *item = (char *)xRingbufferReceiveUpTo(dataBuffer_mesh, &itemSize, portMAX_DELAY, 1024 * 7); 
        _mesh_write(NULL, item, itemSize, portMAX_DELAY, NULL);
        vRingbufferReturnItem(dataBuffer_mesh, (void *)item);
        vTaskDelay( 30 / portTICK_RATE_MS);  
    }
}

uint8_t dst_addr[MWIFI_ADDR_LEN] = MWIFI_ADDR_BROADCAST; 
mwifi_data_type_t data_type = {0};
esp_err_t _mesh_write(audio_element_handle_t self, char *buffer, int len, TickType_t ticks_to_wait, void *context)
{
    data_type.communicate = MWIFI_COMMUNICATE_MULTICAST;

    esp_err_t ret; 
    memcpy(dest.addr, wifi_sta_list.sta[0].mac, 6);
    int retry_count = 3;
    ret = mwifi_root_write(dst_addr, 1, &data_type, buffer, len, true);

    return len;
}

//Child Node task where it receives

static void mesh_receive_buff(void *arg)
{
[... 359 lines cut ...]
W (48584) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
W (48592) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 3
W (48600) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 3
W (48608) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 4
W (48616) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 4
W (48624) [mwifi, 1012]: Part of the packet is lost, expect_seq: 1, recv_seq: 0
W (48632) [mwifi, 1012]: Part of the packet is lost, expect_seq: -1, recv_seq: 1
W (48640) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 1
W (48648) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
W (48656) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
W (48664) [mwifi, 1012]: Part of the packet is lost, expect_seq: 1, recv_seq: 0
W (48672) [mwifi, 1012]: Part of the packet is lost, expect_seq: -1, recv_seq: 1
W (48680) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 1
W (48688) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
W (48696) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
W (48704) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 3
W (48712) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 3
W (48720) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 4
W (48728) [mwifi, 1012]: Part of the packet is lost, expect_seq: 0, recv_seq: 4
W (48736) [mwifi, 1012]: Part of the packet is lost, expect_seq: 1, recv_seq: 0
```

sdkconfig of child nodes

[sdkconfig.txt](https://github.com/espressif/esp-mdf/files/6989037/sdkconfig.txt)

zhanzhaocheng, 2021-08-16:
1. This is due to packet loss caused by too frequent packet sending and too much data volume, which exceeds the bandwidth of MESH
There are several solutions as follows:
1. 7*1024 data is sent in five times, each time only the length of the packet does not exceed `MWIFI_PAYLOAD_LEN` bytes
2. Modify the Wi-Fi configuration of mwifi to increase bandwidth：
[How to configure parameters](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#how-to-configure-parameters)
[image]

ehabulhaq, 2021-10-03:
the solution you suggested didnt really help

The data bandwidth I am trying to send is 1.4 Mbps. isnt that enough for MESH to handle? I thought mesh can support up to10 Mbps