espressif/esp-mdf #239: ESP-MDF Mupgrade procedure question

sw-dev-code, 2021-08-30:
I'm doing the testing of the ESP-MESH network and I'm interested in what will happen in a case when Mupgrade process is started but one of the nodes is connected to the MESH network but not in a range of a ROOT device?

esp-lqq, 2021-09-07:
1. You can refer to the code during the upgrade process.
2. The device upgrade is divided into two steps. The root node downloads the firmware and synchronizes it in the firmware network.
After the firmware download is complete, the mac addresses of all devices in the network that need to be upgraded will be obtained. If the device has joined the network at this time, it can be upgraded by the way, otherwise it cannot be upgraded.

sw-dev-code, 2021-09-07:
Great, for me the most important thing is that edge nodes (not connected to root) is going to be updated.

Thank you for the help.

closed 2021-09-07 by sw-dev-code