espressif/esp-mdf #241: 使用esp-mdf的mesh-with-aliyun示例时，修改设备的三元组不成功

PolarLights-Snow, 2021-09-11:
## （除三元组及WiFi配置外其他代码都没改）编译下载后，串口看到打印的三元组还是之前的内容，帮忙分析下，非常感谢。

## sdkconfig文件内容如下：
[image]

## 编译烧写指令如下：
[image]

## 打印数据如下：
[image]

jorislee, 2021-09-13:
您好，默认会在第一次运行时将配置写入 nvs 中，第二次启动便在 nvs 中读取。
您需要将 flash 全部擦除重新烧录或者重新烧录 nvs 三元组。

PolarLights-Snow, 2021-09-13:
好的，非常感谢。我看到你们有批量修改nvs的工具，但我没找到相关资源，能否给我发一下工具的链接以及使用文档？麻烦了------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年9月13日(星期一) 晚上8:24
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [espressif/esp-mdf] 使用esp-mdf的mesh-with-aliyun示例时，修改设备的三元组不成功 (#241)

PolarLights-Snow, 2021-09-14:
> 你好，默认会在第一次运行时将配置写入 nvs 中，第二次启动便在 nvs 中读取。
> 您需要闪一下全部重重烧录或重烧录 nvs 三元组。

好的，非常感谢。我看到你们有示范修改的工具，但我没有找到相关资源，能否给我发一下工具的链接以及使用文档？麻烦了

esp-lqq, 2021-09-15:
之前我们有构建脚本，用以减少 NVS 生成的繁琐过程。
[multi_nvs_partition_gen](https://github.com/espressif/esp-mdf/blob/master/examples/aliyun_linkkit/tools/multi_nvs_partition_gen.py) 是一个将阿里云三元组 csv 表格转换为可 nvs 的 csv 表格。
[nvs_partition_generator](https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py)
 可以将 csv 转换为可以烧录至 nvs 分区的 bin 文件。

PolarLights-Snow, 2021-09-15:
> 之前我们有构建脚本，用以减少NVS生成的繁琐过程。
> [multi_nvs_partition_gen](https://github.com/espressif/esp-mdf/blob/master/examples/aliyun_linkkit/tools/multi_nvs_partition_gen.py)是一个将阿里云三元组CSV表格转换为可NVS的CSV表格。
> [nvs_partition_generator](https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py)
> 可以将CSV转换为可以烧录至NVS分区的.bin作为文件。

非常感谢，我去看看试一下