espressif/esp-mdf #242: Release Plan

ClaesIvarsson, 2021-09-24:
I would like to know your plans for integrating stable versions of esp-idf into esp-mdf.
Current **release/v1** branch in esp-mdf is using v3.3 of the esp-idf and there is numerous 4.x.y versions released.

Can I expect an esp-mdf release based on esp-idf 4.x.x anytime soon ?

LarsKumbier, 2021-09-27:
To add to the question: What's stability of this project? There was one release 18 months ago and while a lot of work is happening by @EspHuifeng, there does not seem to be a team behind like the ESP-IDF repository. Is this project mature and supported enough at Espressif to be used in production?

LarsKumbier, 2021-09-28:
Another point: ESP-IDF v3.3 (according to the [Support Policy](https://github.com/espressif/esp-idf/blob/master/SUPPORT_POLICY.md)) is already in maintenance mode and will be EOL in 5 months from now. Since the current stable v1.0 is based on IDF v3.3, I suspect that the next version of ESP-MDF will be a major release as well, since the current master is referencing ESP-IDF v4.3? While the work needed to move from ESP-IDF v3 to v4 is [documented](https://github.com/espressif/esp-idf/releases/tag/v4.0), it's unclear how much work is needed between ESP-MDF v1.0 and the next release.

All of these points are big risks for starting a new project on this framework. Could you clarify the points, so that we can make an informed decision to either start with ESP-MDF or use ESP-IDF instead? 

Thank you!

zhanzhaocheng, 2021-09-29:
ESP-MDF is adapted to esp-idf v4.3.1, currently in internal testing, it is expected to be updated to github in mid-October

zhanzhaocheng, 2021-10-18:
The adaptation based on v4.3.1 has been updated to the master branch. We have tested it internally and can be used in production. If there are any problems, we will fix them in time
The branch release/v2.0 will be checkouted recently, and its support periods will follow the support plan of [esp-idf v4.x](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#support-periods)

ClaesIvarsson, 2021-11-03:
We passed end of October now and I would like to see an update on the progress of the migration to IDF 4.x.
Can you please inform when it is expected to be ready ?

rmshub, 2021-11-17:
Any update on esp-mdf release/v2.0 to support IDF 4.x ?

ClaesIvarsson, 2021-11-23:
Can we get an update on this topic ?

eos1d3, 2022-01-19:
It seems MDF is dying. And whether it can be used for production is really a big question. If Espressif does not officially promise its future, I think it is better to skip it.

Jiangyafeng, 2022-02-17:
Hi All,

Sorry, due to the internal staff transfer, the maintenance of this mdf warehouse has been delayed, and maintenance and updates will still be carried out in the future.

For the launch of mdf release/v2.0, this requires a long time to test its stability. It will take more than two months of testing, and a stable release/v2.0 version will be launched by then.

Jiangyafeng, 2022-04-19:
Hi @vitorrussi 
Sorry that I can not finish the test in time due to the recent COVID-19 outbreak and now I am in home quarantine.
As soon as work officially resumes in Shanghai, China, I will deal with this as soon as possible and update the latest version in time.

PatrickV82, 2022-07-14:
@Jiangyafeng any updates?