espressif/esp-mdf #244: win10环境搭建问题

Mr-Cyclone, 2021-09-28:
1.已有IDF的环境下，如何搭建MDF环境
2.git clone --recursive命令失败，并且使用esp-gitee-tools重定向子仓库下载也失败如何解决
3.翻墙拉取代码同样失败
4.在wsl环境中拉取代码也是同上的问题

ps：idf照着文档很好操作，到了mdf各种问题。。。对菜鸟很不友好，请大神帮忙

zhanzhaocheng, 2021-09-30:
推荐你参考这个视频看是否对你有帮助：[从零开始带你完成 Windows 系统下 ESP-MDF 环境搭建！MESH 开发必备](https://www.bilibili.com/video/BV1GT4y1A75N?spm_id_from=333.999.0.0)

Mr-Cyclone, 2021-09-30:
> 推荐你参考这个视频看是否对你有帮助：[从零开始带你完成 Windows 系统下 ESP-MDF 环境搭建！MESH 开发必备](https://www.bilibili.com/video/BV1GT4y1A75N?spm_id_from=333.999.0.0)

抱歉，网上既有的资料我都看过了包括这个视频，解决不了问题

zhanzhaocheng, 2021-09-30:
`git clone git@github.com:espressif/esp-mdf.git` 不带 --recursive，然后修改 esp-idf 的路径配置 https://github.com/espressif/esp-mdf/blob/master/project.cmake#L15