espressif/esp-mdf #257: Which IDF version to use

ClaesIvarsson, 2021-11-03:
I am looking at the master branch of esp-mdf and can't see that esp-idf isn't a submodule anymore.

Can you please explain how to know which version of esp-idf to use for esp-mdf ?

vitorrussi, 2021-11-03:
[README Quick-Start](https://github.com/espressif/esp-mdf#quick-start) shows which ESP-IDF version you should install:

> 1. [**Setup Toolchain**](https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/index.html#step-1-set-up-the-toolchain): please set up according to your PC's operating system ([Windows](https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/windows-setup.html), [Linux](https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/linux-setup.html) or [Mac OS](https://docs.espressif.com/projects/esp-idf/en/stable/get-started-cmake/macos-setup.html)). If you use linux, you can use this commands.
> 
>     ```shell
>     git clone -b v4.3.1 --recursive https://github.com/espressif/esp-idf.git
>     cd ~/esp/esp-idf
>     ./install.sh
>     . ./export.sh
>     ```

ClaesIvarsson, 2021-11-04:
> [README Quick-Start](https://github.com/espressif/esp-mdf#quick-start) shows which ESP-IDF version you should install:

Ok, my bad, should have read the documentation maybe.

I have been using esp-mdf on master branch for a long time and esp-idf has been a submodule during this time.
Anyone know why this has changed ?

yiziliuxian, 2021-11-04:
I guess the current way will be easier to switch between different esp-idf verisons.

zhanzhaocheng, 2021-11-05:
As @yiziliuxian  said, in order to better switch between different esp-idf and simplify the environment construction process
We will modify the document to identify this change in a more conspicuous manner and mark the version that esp-idf has been verified and tested.

ClaesIvarsson, 2021-11-05:
> As @yiziliuxian said, in order to better switch between different esp-idf and simplify the environment construction process We will modify the document to identify this change in a more conspicuous manner and mark the version that esp-idf has been verified and tested.

I realize that you can have different strategies on this.
* no question which version has been verified and tested and should be used if it is a submodule.
* more freedom to select if it is NOT a submodule

As a user of the esp-mdf I would prefer submodule

ClaesIvarsson, 2021-11-12:
I am using esp-mdf cf502740f5a6c82a0dc3059e7591c262795f70b2 together with espressif/esp-idf@2e74914051d14ec2290fc751a8486fb51d73a31e

I keep getting compiler errors saying fatal error: hal/cpu_hal.h: No such file or directory
When I check include folders in the compiler command I can't see why this happens as paths looks ok.

Can someone confirm that the two hashes actually should work together ?
Would appreciate if someone can share some knowledge about this problem.

Jiangyafeng, 2022-02-17:
@ClaesIvarsson  
The current architecture of MDF is like this, the IDF version corresponding to mdf release/v1.0 is release/v3.3, and it can be automatically downloaded as a submodule through git submodule update --init --recursive.

The IDF version corresponding to the mdf mater branch is release/v4.3.1 . This version does not automatically download its IDF library as a submodule, and needs to be downloaded separately.

And for the compilation errors you have now, is it convenient to send the corresponding log? This will better solve your problem.

vitorrussi, 2022-02-17:
@Jiangyafeng does MDF master branch works with IDF 4.4?

ClaesIvarsson, 2022-02-17:
> @ClaesIvarsson The current architecture of MDF is like this, the IDF version corresponding to mdf release/v1.0 is release/v3.3, and it can be automatically downloaded as a submodule through git [...]

Thanks for the clarification BUT I am NOT able to find a version of IDF that compiles with MDF cf502740f5a6c82a0dc3059e7591c262795f70b2 or newer on the masetr branch.
I feel that the release/1.0 of MDF is too far behind on IDF for me to use, and now I would like to update to latest on MDF master branch to check the status. However, after you decision to disconnect the IDF repo I am not able to find a corresponding version if IDF that event compiles.

Please see my [comment ](https://github.com/espressif/esp-mdf/issues/257#issuecomment-966874612) from nov 12 above.
I see a big disadvantage as it is hard to know version combinations that you actually are using and testing. Referencing barnch names doesn't seem to be good enough.

mmrein, 2022-02-18:
I've run some basic tests on master repo with IDF v4.3.2 and v4.4, compiling without problems. Using project which was originally created with older master with integrated IDF, which was v4.1 or v4.2 at the time.

ClaesIvarsson, 2022-02-18:
I have found the reason for the problems I faced migrating to newer version of MDF/IDF.
The reason my project didn't build was that there was a conflict in component names between IDF and my project.
Both project defined a component named *hal*. 

I have renamed my component  and managed to fix all other code related migrations. The project is no building fine.

ClaesIvarsson, 2023-05-18:
I have skipped the mdf completely now. Only using the mupgrade component, copied into my own repo.
This way I no longer need to keep track of the relationship between mdf and idf.

closed 2023-05-18 by ClaesIvarsson

tonyshuang, 2023-05-18:
您好，信件已收到。