espressif/esp-mdf #258: Node节点运行一段时间后，会掉网。

llltong, 2021-11-10:
请问Node节点运行一段时间后，会掉网，然后就一直无法重新加回原先的网络中，然后这台设备就永远的在不断的请求网络。使用的是router这个工程；
[Mesh4Relay, 1521]: Node send, size: 40, data: 0102030405060819,Mesh4Relay,0,1.1.0,-30,[0m                              ### 正常的数据
D (1156537) [Mesh4Relay, 1521]: Node send, size: 40, data: 0102030405060819,Mesh4Relay,0,1.1.0,-32,[0m         ### 正常的数据
D (1157545) [Mesh4Relay, 1521]: Node send, size: 40, data: 0102030405060819,Mesh4Relay,0,1.1.0,-31,[0m         ### 正常的数据
D (1158549) [Mesh4Relay, 1521]: Node send, size: 40, data: 0102030405060819,Mesh4Relay,0,1.1.0,-31,[0m         ### 正常的数据
这里开始出现问题。
W (1158854) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:2, 300 ms timeout, seqno:0, xseqno:1125, no_wnd_count:0, timeout_count:0
W (1159155) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:2, 300 ms timeout, seqno:0, xseqno:1125, no_wnd_count:0, timeout_count:1
W (1159456) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:2, 300 ms timeout, seqno:0, xseqno:1125, no_wnd_count:0, timeout_count:2
W (1159757) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:2, 300 ms timeout, seqno:0, xseqno:1125, no_wnd_count:0, timeout_count:3
W (1160058) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:2, 300 ms timeout, seqno:0, xseqno:1125, no_wnd_count:0, timeout_count:4
........................同样的请求
W (1169580) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:91
W (1169591) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (1169881) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:92
I (1169923) wifi:bcn_timout,ap_probe_send_start
I (1169924) wifi:pm stop, total sleep time: 924633195 us / 1153939460 us

I (1169924) wifi:(mesh)pm stop, sleep:881 secs, total:1100 secs, current active set:10, running:0.80
W (1170182) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:93
W (1170484) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:94
W (1170785) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:95
W (1171086) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:96
W (1171388) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:97
W (1171689) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:98
W (1171990) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:99
W (1172291) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:100
I (1172438) wifi:ap_probe_send over, resett wifi status to disassoc
I (1172439) wifi:state: run -> init (c800)
I (1172440) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
[0;32mI (1172446) [mwifi, 188]: Parent is disconnected, reason: 200[0m
[0;32mI (1172450) [Mesh4Relay, 1612]: event_loop_cb, event: 8[0m
I (1172451) mesh: [wifi]disconnected reason:200(beacon timeout), continuous:1/max:12, non-root, vote(,stopped)<><>
[0;32mI (1172457) [Mesh4Relay, 1665]: Parent is disconnected on station interface[0m
I (1172471) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1172481) wifi:state: init -> auth (b0)
W (1172593) mesh: [mesh_schedule.c,3081] [WND-RX]max_wnd:4, 300 ms timeout, seqno:0, xseqno:1107, no_wnd_count:0, timeout_count:101
[0;33mW (1172594) [mwifi, 686]: <ESP_ERR_MESH_DISCONNECTED> Node failed to send packets, dest_addr: ff:00:00:01:00:00, flag: 0x28, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffff8b0, size: 40[0m
[0;33mW (1172612) [mwifi, 939]: <ESP_ERR_MESH_DISCONNECTED> Node failed to send packets, data_flag: 0x28, dest_mac: ff:00:00:01:00:00[0m
[0;33mW (1172624) [Mesh4Relay, 1527]: <ESP_ERR_MESH_DISCONNECTED> mwifi_write[0m
I (1173984) wifi:state: auth -> init (200)
I (1173985) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
[0;32mI (1173988) [mwifi, 188]: Parent is disconnected, reason: 2[0m
I (1173990) mesh: [wifi]disconnected reason:2(auth expire), continuous:2/max:12, non-root, vote(,stopped)<><>
[0;32mI (1173991) [Mesh4Relay, 1612]: event_loop_cb, event: 8[0m
I (1174004) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1174012) wifi:state: init -> auth (b0)
[0;32mI (1174016) [Mesh4Relay, 1665]: Parent is disconnected on station interface[0m
I (1175516) wifi:state: auth -> init (200)
I (1175517) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
[0;32mI (1175519) [mwifi, 188]: Parent is disconnected, reason: 2[0m
[0;32mI (1175521) [Mesh4Relay, 1612]: event_loop_cb, event: 8[0m
I (1175522) mesh: [wifi]disconnected reason:2(auth expire), continuous:3/max:12, non-root, vote(,stopped)<><>
[0;32mI (1175528) [Mesh4Relay, 1665]: Parent is disconnected on station interface[0m
I (1175542) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1175551) wifi:state: init -> auth (b0)
I (1177055) wifi:state: auth -> init (200)
I (1177056) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
[0;32mI (1177058) [mwifi, 188]: Parent is disconnected, reason: 2[0m
[0;32mI (1177060) [Mesh4Relay, 1612]: event_loop_cb, event: 8[0m
I (1177061) mesh: [wifi]disconnected reason:2(auth expire), continuous:4/max:12, non-root, vote(,stopped)<><>
[0;32mI (1177067) [Mesh4Relay, 1665]: Parent is disconnected on station interface[0m
I (1177081) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1177091) wifi:state: init -> auth (b0)
I (1178594) wifi:state: auth -> init (200)
I (1178595) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
[0;32mI (1178598) [mwifi, 188]: Parent is disconnected, reason: 2[0m
[0;32mI (1178600) [Mesh4Relay, 1612]: event_loop_cb, event: 8[0m
[0;32mI (1178680) [Mesh4Relay, 1665]: Parent is disconnected on station interface[0m
[0;32mI (1178682) [mwifi, 266]: Network state: root_less[0m
[0;32mI (1178684) [Mesh4Relay, 1612]: event_loop_cb, event: 11[0m
I (1178686) mesh: 5180[healing]looking for a new parent, [L:2]try layer:1[revote][scan]
[0;32mI (1178691) [Mesh4Relay, 1612]: event_loop_cb, event: 20[0m
W (1179301) wifi:scan number is 0
I (1179301) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1179305) mesh: [FAIL][1]root:0, fail:1, normal:0, <pre>backoff:0

W (1179912) wifi:scan number is 0
I (1179913) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1179916) mesh: [FAIL][2]root:0, fail:2, normal:0, <pre>backoff:0

W (1180524) wifi:scan number is 0
I (1180524) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1180528) mesh: [FAIL][3]root:0, fail:3, normal:0, <pre>backoff:0

W (1180656) wifi:scan number is 0
I (1180656) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1180660) mesh: [FAIL][4]root:0, fail:4, normal:0, <pre>backoff:0

W (1181267) wifi:scan number is 0
I (1181268) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1181271) mesh: [FAIL][5]root:0, fail:5, normal:0, <pre>backoff:0

W (1181878) wifi:scan number is 0
I (1181879) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1181882) mesh: [FAIL][6]root:0, fail:6, normal:0, <pre>backoff:0

W (1182010) wifi:scan number is 0
I (1182011) mesh: [SCAN][ch:1]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][06:05:88:93:72:eb]<>
I (1182014) mesh: [FAIL][7]root:0, fail:7, normal:0, <pre>backoff:0

llltong, 2021-11-10:
发现他掉网了以后，在这个if (!mwifi_is_connected())中一直打印error，有什么方法让设备重新连接上原来的网络吗？
static void node_write_task(void *arg)
{
    size_t size                     = 0;
    char *data                      = NULL;
    mdf_err_t ret                   = MDF_OK;
    mwifi_data_type_t data_type     = {0};

    MDF_LOGI("NODE task is running");

    for (;;) {
        //if (!mwifi_is_connected() && !(mwifi_is_started() && esp_mesh_is_root())) {
        if (!mwifi_is_connected()) {
            vTaskDelay(500 / portTICK_RATE_MS);
           MDF_LOGI("error");//打印错误
            continue;
        }
        //组建要发送的数据
        size = asprintf(&data,"%s,%s,%x,%s,%d,",denine_devid,dev_name,Relay_state,DEVVERSION,mwifi_get_parent_rssi());
        data_type.custom = 0x01;
        MDF_LOGD("Node send, size: %d, data: %s", size, data);
        ret = mwifi_write(NULL, &data_type, data, size, true);
        MDF_FREE(data);
        MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_write", mdf_err_to_name(ret));

        vTaskDelay(1000 / portTICK_RATE_MS);//1S定时发送
    }

    MDF_FREE(data);
    MDF_LOGW("NODE task is exit");

    vTaskDelete(NULL);
}

tomyqg, 2021-11-13:
我遇到的是root重连了一个路由器，但是node还一直在原来的信道上扫描(node root 都使能了信道切换，root还使能了路由切换，root 是esp32 node是esp32c3，idf 和 mdf均为master版本)，会一直set ROOTLESS模式，连接不上root了

zhanzhaocheng, 2021-11-15:
@llltong 在 `mwifi_start()` 之后调用 [esp_mesh_send_block_time()](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_mesh.h#L696-L706) 设置发包超时时间，可以解决 WND 满造成组网失败的问题：
```c

```

llltong, 2021-11-16:
> @llltong 在 `mwifi_start()` 之后调用 [esp_mesh_send_block_time()](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_mesh.h#L696-L706) 设置发包超时时间，可以解决 WND 满造成组网失败的问题：

好的，谢谢，请问下我是把这个时间设置大一点就好还是小一点， 我看那个报错是300ms的，那我设置成500ms，例如esp_mesh_send_block_time(500)，这样是不是就解决这个问题了？