espressif/esp-mdf #259: ROOT调用sendto不报错，但是数据没有传输到PC (PC和ROOT连接同一个AP)

tomyqg, 2021-11-11:
我有两个模块，一个固定为root，另一个为node，它们之间通过mesh 广播交互，同时ROOT和PC建立udp通信，将node发送的数据上传到PC上，目前PC->ROOT->NODE通信是正常的，但是NODE->ROOT（ROOT能收到NODE返回的数据）->PC时经常会出现ROOT 通过 sendto 发送成功，但是PC收不到数据的问题，请问这个问题该如何排查，谢谢！

llltong, 2021-11-11:
node结节发送的函数不是 sendto吧。

tomyqg, 2021-11-12:
类似的问题18年就有人问了 [https://esp32.com/viewtopic.php?t=7866](url)， 不知目前是什么状态啊，是否无解？

llltong, 2021-11-12:
> 类似的问题18年就有人问了 [https://esp32.com/viewtopic.php?t=7866](url)， 不知目前是什么状态啊，是否无解？

你用的是哪个工程改的？NODE节点发送数据给ROOT，ROOT在通过UDP转发给PC机，我这边是正常的，已经运行在产品上了，工作还是稳定的。

tomyqg, 2021-11-12:
用router 那个改的， 我是PC通过root给node发指令，node的应答到了root这里偶尔会丢失（sendto没有报错，但PC收不到应答），你有啥特殊配置吗？

llltong, 2021-11-12:
没有特殊配置，估计是你代码的问题了

tomyqg, 2021-11-12:
关键它不报错，就让人很郁闷

https://github.com/espressif/esp-mdf/issues/259#issuecomment-966980713

closed 2021-12-05 by tomyqg