espressif/esp-mdf #260: esp32s2 wifi mesh组网

Zhangtingyou1, 2021-11-12:
esp-mdf  WiFi mesh 组网发现，连接路由器组网连接成功后，直接断掉所连接路由器电源，会大概率出现cache error错误无限重启，小概率出现无法重连到路由器但是mesh组网正常，就是一直连接不上路由器，经测试路由器必须断掉电源bug才会复现，重启和其他操作不会复现bug，附件图片和文档为MDF与IDF版本与出现错误的串口信息

Zhangtingyou1, 2021-11-12:
~/esp-mdf/esp-idf$ git log
commit e9cf9e2978d0187b0f83a9978391217e39946517 (HEAD, tag: v4.3-beta3)
Merge: 6463dd9630 e16ec9a574
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Wed Apr 14 16:23:57 2021 +0000

    Merge branch 'bugfix/e8192_and_assert_param_bt1551' into 'release/v4.3'
    
    Modify E8192 ELx 200 ELx40 log level to LOGD
    
    See merge request espressif/esp-idf!12977

Zhangtingyou1, 2021-11-12:
/esp-mdf/esp-idf$ git branch -v
* (HEAD detached at e9cf9e2978) e9cf9e2978 Merge branch 'bugfix/e8192_and_assert_param_bt1551' into 'release/v4.3'
  release/v4.3                  47b96db12d [behind 427] Merge branch 'feature/prov_mgr_reset_state_v4.3' into 'release/v4.3'

Zhangtingyou1, 2021-11-12:
~/esp-mdf$ git branch -v
* master 0b59603 [behind 9] Merge branch 'bugfix/addr_verification_info_error' into 'master'
  v1.0   a68cf42 [behind 2] Merge branch 'bugfix/distribution_network' into 'release/v1.0'

Zhangtingyou1, 2021-11-12:
无限重启串口打印信息：
I (12947) mesh: 5136<active>parent layer:1(node), channel:13, rssi:-38, assoc:2(cnx rssi threshold:-120)my_assoc:0
D (14034) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
I (14451) mesh: 5927<scan>parent layer:1, rssi:-38, assoc:2(cnx rssi threshold:-120)
I (14451) mesh: [SCAN][ch:13]AP:2, other(ID:0, RD:0), MAP:2, idle:0, candidate:2, root:1, topMAP:0[c:2,i:2][4c:77:66:d2:4f:17]<weak>
I (14459) mesh: 7349[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (14467) mesh: 701[monitor]no change, parent:7c:df:a1:0e:56:bd, rssi:-38
I (14473) mesh: 1989<arm>parent monitor, my layer:2(cap:16)(node), interval:4400ms, retries:2<>
D (17049) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
I (18988) mesh: 5136<active>parent layer:1(node), channel:13, rssi:-38, assoc:2(cnx rssi threshold:-120)my_assoc:0
D (20053) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
I (20491) mesh: 5927<scan>parent layer:1, rssi:-37, assoc:2(cnx rssi threshold:-120)
I (20492) mesh: [SCAN][ch:13]AP:2, other(ID:0, RD:0), MAP:2, idle:0, candidate:2, root:1, topMAP:1[c:0,i:2][4c:77:66:d2:4f:17]<weak>
I (20500) mesh: 7349[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (20507) mesh: 701[monitor]no change, parent:7c:df:a1:0e:56:bd, rssi:-37
I (20514) mesh: 1989<arm>parent monitor, my layer:2(cap:16)(node), interval:7338ms, retries:3<>
[0;32mI (22024) [mqtt_examples, 183]: System information, channel: 13, layer: 2, self mac: 7c:df:a1:32:92:98, parent bssid: 7c:df:a1:0e:56:bd, parent rssi: -39, node num: 3, free heap: 74772[0m
[0;32mI (22031) [mdf_mem, 122]: (827) <mwifi: 301> ptr: 0x3ffdeaf0, size: 32[0m
[0;32mI (22038) [mdf_mem, 122]: (827) <mwifi: 306> ptr: 0x3ffdf578, size: 176[0m
[0;32mI (22045) [mdf_mem, 122]: (2018) <mqtt_examples: 100> ptr: 0x3ffed178, size: 1456[0m
[0;32mI (22053) [mdf_mem, 122]: (8024) <mwifi: 979> ptr: 0x3ffd0f98, size: 1456[0m
[0;32mI (22061) [mdf_mem, 129]: Memory record, num: 4, size: 3120[0m
D (23064) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
D (26068) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
I (28037) mesh: 5136<active>parent layer:1(node), channel:13, rssi:-38, assoc:2(cnx rssi threshold:-120)my_assoc:0
D (29079) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
I (29540) mesh: 5927<scan>parent layer:1, rssi:-36, assoc:2(cnx rssi threshold:-120)
I (29541) mesh: [SCAN][ch:13]AP:2, other(ID:0, RD:0), MAP:2, idle:0, candidate:2, root:1, topMAP:1[c:0,i:2][4c:77:66:d2:4f:17]<weak>
I (29549) mesh: 7349[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (29556) mesh: 701[monitor]no change, parent:7c:df:a1:0e:56:bd, rssi:-36
I (29563) mesh: 1989<arm>parent monitor, my layer:2(cap:16)(node), interval:394886ms, retries:3<>
[0;32mI (32024) [mqtt_examples, 183]: System information, channel: 13, layer: 2, self mac: 7c:df:a1:32:92:98, parent bssid: 7c:df:a1:0e:56:bd, parent rssi: -46, node num: 3, free heap: 74780[0m
[0;32mI (32031) [mdf_mem, 122]: (827) <mwifi: 301> ptr: 0x3ffdeaf0, size: 32[0m
[0;32mI (32038) [mdf_mem, 122]: (827) <mwifi: 306> ptr: 0x3ffdf578, size: 176[0m
[0;32mI (32045) [mdf_mem, 122]: (2018) <mqtt_examples: 100> ptr: 0x3ffed178, size: 1456[0m
[0;32mI (32053) [mdf_mem, 122]: (8024) <mwifi: 979> ptr: 0x3ffd0f98, size: 1456[0m
[0;32mI (32061) [mdf_mem, 129]: Memory record, num: 4, size: 3120[0m
D (32084) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
D (35096) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
D (38108) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
D (41117) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
[0;32mI (42024) [mqtt_examples, 183]: System information, channel: 13, layer: 2, self mac: 7c:df:a1:32:92:98, parent bssid: 7c:df:a1:0e:56:bd, parent rssi: -43, node num: 3, free heap: 74780[0m
[0;32mI (42031) [mdf_mem, 122]: (827) <mwifi: 301> ptr: 0x3ffdeaf0, size: 32[0m
[0;32mI (42038) [mdf_mem, 122]: (827) <mwifi: 306> ptr: 0x3ffdf578, size: 176[0m
[0;32mI (42045) [mdf_mem, 122]: (2018) <mqtt_examples: 100> ptr: 0x3ffed178, size: 1456[0m
[0;32mI (42053) [mdf_mem, 122]: (8024) <mwifi: 979> ptr: 0x3ffd0f98, size: 1456[0m
[0;32mI (42061) [mdf_mem, 129]: Memory record, num: 4, size: 3120[0m
D (44121) [mqtt_examples, 151]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1329298", "parent":"7cdfa10e56bd","layer":2}[0m
I (44353) mesh: <MESH_NWK_MIE_CHANGE><><TODS><><><><><>
[0;32mI (44354) [mwifi, 258]: State represents: 1[0m
[0;32mI (44354) [mqtt_examples, 236]: event_loop_cb, event: 11[0m
[... cut ...]
Guru Meditation Error: Core  0 panic'ed (Cache error). 

Core  0 register dump:
PC      : 0x40140b89  PS      : 0x00060d34  A0      : 0x80024c49  A1      : 0x3fffe280  
A2      : 0x02000000  A3      : 0x00000000  A4      : 0x02000000  A5      : 0x00000004  
A6      : 0x000000c0  A7      : 0x00000000  A8      : 0x3ffffd70  A9      : 0x3fffe240  
A10     : 0x00fdffff  A11     : 0x00000030  A12     : 0x00000000  A13     : 0x50000008  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000007

zhanzhaocheng, 2021-11-16:
@Zhangtingyou1 此问题可能是由于 [memprot](https://github.com/espressif/esp-idf/commit/c990c8f7521ef203ef1d1a35ad8dea60da56e343) 的 bug 引发的，请更新 esp-idf 验证：
```shell
cd esp-idf
git pull
git checkout release/v4.3
git submodule update --init --recursive
```

Zhangtingyou1, 2021-11-16:
好的

Zhangtingyou1, 2021-11-17:
更新后测试，断掉wifi电源不会出现无限重启，但是会重启一次报错Guru Meditation Error: Core  0 panic'ed (Memory protection fault).

Zhangtingyou1, 2021-11-17:
抓取日志：
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWEROhannel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (8082) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:3
]fail to find a network, cI (12001) mesh: [FIND][ch:0]AP:20, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (12002) mesh: [FIND:3]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (12009) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:4
W (12600) wifi:Haven't to connect to a suitable AP now!
[0;32mI (12600) [mqtt_examples, 183]: System information, channel: 5, layer: -1, self mac: 7c:df:a1:0e:5c:38, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 0, free heap: 81232[0m
[0;32mI (12612) [mdf_mem, 122]: (660) <mwifi: 301> ptr: 0x3ffde638, size: 32[0m
[0;32mI (12618) [mdf_mem, 122]: (660) <mwifi: 306> ptr: 0x3ffdf0c0, size: 176[0m
[0;32mI (12626) [mdf_mem, 122]: (2594) <mqtt_examples: 100> ptr: 0x3ffecbd8, size: 1456[0m
[0;32mI (12634) [mdf_mem, 129]: Memory record, num: 3, size: 1664[0m
I (13594) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (13594) mesh: [FIND:4]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (13603) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:5
I (17521) mesh: [FIND][ch:0]AP:15, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (17521) mesh: [FIND:5]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (17529) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:6
I (19112) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (19113) mesh: [FIND:6]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (19122) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:7
W (22600) wifi:Haven't to connect to a suitable AP now!
[0;32mI (22600) [mqtt_examples, 183]: System information, channel: 12, layer: -1, self mac: 7c:df:a1:0e:5c:38, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 0, free heap: 80176[0m
[0;32mI (22612) [mdf_mem, 122]: (660) <mwifi: 301> ptr: 0x3ffde638, size: 32[0m
[0;32mI (22619) [mdf_mem, 122]: (660) <mwifi: 306> ptr: 0x3ffdf0c0, size: 176[0m
[0;32mI (22626) [mdf_mem, 122]: (2594) <mqtt_examples: 100> ptr: 0x3ffecbd8, size: 1456[0m
[0;32mI (22634) [mdf_mem, 129]: Memory record, num: 3, size: 1664[0m
I (23039) mesh: [FIND][ch:0]AP:11, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (23039) mesh: [FIND:7]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (23047) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:8
I (24631) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (24631) mesh: [FIND:8]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (24640) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:9
I (28559) mesh: [FIND][ch:0]AP:24, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (28559) mesh: [FIND:9]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (28567) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:10
I (30151) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (30151) mesh: [FIND:10]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (30160) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:11
W (32600) wifi:Haven't to connect to a suitable AP now!
[0;32mI (32600) [mqtt_examples, 183]: System information, channel: 9, layer: -1, self mac: 7c:df:a1:0e:5c:38, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 0, free heap: 79984[0m
[0;32mI (32612) [mdf_mem, 122]: (660) <mwifi: 301> ptr: 0x3ffde638, size: 32[0m
[0;32mI (32618) [mdf_mem, 122]: (660) <mwifi: 306> ptr: 0x3ffdf0c0, size: 176[0m
[0;32mI (32626) [mdf_mem, 122]: (2594) <mqtt_examples: 100> ptr: 0x3ffecbd8, size: 1456[0m
[0;32mI (32634) [mdf_mem, 129]: Memory record, num: 3, size: 1664[0m
I (34078) mesh: [FIND][ch:0]AP:23, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (34078) mesh: [FIND:11]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (34086) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:12
I (35670) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (35670) mesh: [FIND:12]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

I (35679) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:13
I (39597) mesh: [FIND][ch:0]AP:17, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]
I (39597) mesh: [FIND:13]fail to find a network, channel:0, cfg<channel:0, router:wifi_test, 00:00:00:00:00:00>

[... cut ...]
[0;32mI (1054) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08[0m
I (1141) wifi:mode : sta (7c:df:a1:0e:5c:38)
I (1142) wifi:enable tsf
[0;32mI (1143) [mwifi, 298]: esp-mdf version: v1.0-147-g0b59603-dirty[0m
I (1145) wifi:mode : sta (7c:df:a1:0e:5c:38) + softAP (7c:df:a1:0e:5c:39)
I (1153) wifi:Total power save buffer number: 16
I (1154) wifi:Init max length of beacon: 752/752
I (1159) wifi:Init max length of beacon: 752/752
I (1167) mesh: <nvs>read layer:1
I (1167) mesh: <nvs>read assoc:2
I (1175) wifi:Total power save buffer number: 16