espressif/esp-mdf #265: 关于ESP32-MeshKit-Light连接 ESP-Mesh App的问题

PolarLights-Snow, 2021-11-26:
在使用说明->[设备应用](https://github.com/espressif/esp-mdf/blob/master/examples/development_kit/light/README_cn.md#3%E8%AE%BE%E5%A4%87%E5%BA%94%E7%94%A8)->3.设备配网中提到：进入 Mesh App, 点击"我的"选项栏中的"云端"按钮。App 切换为云端控制, 登录账号后,点击右上角+按钮,点击添加设备.
可是实际应用中，APP一旦切换为云端控制，用户位置将会显示未登录，且没有登陆入口，且不能添加设备。如此我就失去了唯一的设备控制端。
请问这个问题是我操作出错还是APP本身问题？有没有解决办法？

zhanzhaocheng, 2021-12-02:
当前支持 阿里云端控制，具体使用方法参见：[ESP32-MeshKit-Light with Aliyun Linkkit](https://github.com/espressif/esp-mdf/blob/master/examples/development_kit/light/README_cn.md#esp32-meshkit-light-with-aliyun-linkkit)

PolarLights-Snow, 2021-12-06:
> 当前支持 阿里云端控制，具体使用方法参见：[ESP32-MeshKit-Light with Aliyun Linkkit](https://github.com/espressif/esp-mdf/blob/master/examples/development_kit/light/README_cn.md#esp32-meshkit-light-with-aliyun-linkkit)

你好，我就是按照这个教程走的，阿里生活物联上已经有设备上线，数据上传下发也没问题。就是App控制有问题。
教程上说：“esp网络套件”App要切换到云端模式。我的操作如下图
[image]

在教程里，演示的App切换后是有用户登录的。可是我显示的是“未登录”。切换后界面如下图。
[image]

然后我点击用户，就没反应，点击添加设备也没有任何反应。请问这是怎么回事？