espressif/esp-mdf #266: Mconfig在ethernet下的问题

huexpub, 2021-11-26:
在Root配置为eth后，载入mconfig 链式配网，node节点能接收到消息，但直接溢出，取消root ethernet后，node节点则可以通过链式配网完成设置。该部分是否只能指定root使用wifi接入互联网？

huexpub, 2021-11-26:
func: mesh_node_mconfig_init
expression: mconfig_queue_read(&mconfig_data, portMAX_DELAY)

abort() was called at PC 0x40384a2d on core 0
Core  0 register dump:
MEPC    : 0x40385a4e  RA      : 0x403861c4  SP      : 0x3fcaf4f0  GP      : 0x3fc8ee00  
TP      : 0x3fc7d49c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x3fcaf554  A0      : 0x3fcaf51c  A1      : 0x3fcaf552  
A2      : 0x00000000  A3      : 0x3fcaf549  A4      : 0x00000001  A5      : 0x3fc98000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcaf4f0: 0x3c0eeb44 0x40384a2c 0x3fcaf550 0x4038c7a0 0x00000039 0x3c0ef810 0x3fcc06e0 0x3fc90fc4
3fcaf510: 0x3fcaf554 0x3fc90fe0 0x3fcaf550 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fcaf530: 0x43502074 0x34783020 0x34383330 0x20643261 0x63206e6f 0x2065726f 0x00000030 0x00000000
3fcaf550: 0x00000030 0x38333034 0x64326134 0x00000000 0x00000000 0x00000000 0x00000000 0x40384a30
3fcaf570: 0x00000000 0x00000000 0x00000000 0x42009b4e 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf590: 0x00000000 0x00000000 0x00000000 0x40389758 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf5b0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0x010000ff
3fcaf5d0: 0x00000000 0xffffffff 0x0000ffff 0xffffffff 0x0000feff 0x000000ff 0x7f7fffff 0x00000000
3fcaf5f0: 0x3fe00000 0x00000000 0x3ff00000 0x00000000 0x40000000 0x94a03595 0x3fdfffff 0x35afe535
3fcaf610: 0x3fe00000 0x94a03595 0x3fcfffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000
3fcaf630: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fcaf650: 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000 0x00000000
3fcaf670: 0x7ff80000 0x7fc00000 0x3fc8e600 0x000033e8 0xa5a5a5a5 0xa5a5a5a5 0x00000160 0x3fcaf400
3fcaf690: 0x3fc98174 0x3fc92298 0x3fc92298 0x3fcaf68c 0x3fc92290 0x00000015 0x3fcb41ac 0x3fcb41ac
3fcaf6b0: 0x3fcaf68c 0x00000000 0x00000004 0x3fcae688 0x77464449 0x696c626c 0x00006b6e 0x00000001
3fcaf6d0: 0x00000000 0x3fcaf680 0x0000000c 0x600c0000 0x00000004 0x00000000 0x3fcafb08 0x42000252
3fcaf6f0: 0x00006ea2 0x00000000 0x3fc9898c 0x3fc989f4 0x3fc98a5c 0x00000000 0x00000000 0x00000001
3fcaf710: 0x00000000 0x00000000 0x00000000 0x420ba670 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf790: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf7b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaf7d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca0000 0x00000014
3fcaf7f0: 0x3fcad79c 0x6f636d04 0x6769666e 0x6168635f 0x3f006e69 0x0000000c 0x3fcaf7f0 0x47434d04
3fcaf810: 0x3fcaf800 0x00000028 0x00000000 0x00000000 0x3fcaf824 0xffffffff 0x3fcaf824 0x3fcaf824
3fcaf830: 0x7ff80000 0x7fc00000 0x3fc8e600 0x3fcaf810 0x00000070 0x3fcaf8a0 0x3fcaf8a0 0x3fcaf8b4
3fcaf850: 0x3fcaf8b0 0x00000000 0x3fcaf85c 0xffffffff 0x3fcaf85c 0x3fcaf85c 0x00000000 0x3fcaf870
3fcaf870: 0xffffffff 0x3fcaf870 0x3fcaf870 0x00000000 0x00000005 0x00000004 0x6900ffff 0x00000000
3fcaf890: 0x00000000 0x00000000 0x3fcaf840 0x0000000c 0x00000000 0x00000004 0x00000000 0x3fcad314
3fcaf8b0: 0x3fcaf83c 0x00000070 0x3fcaf914 0x3fcaf914 0x3fcaf928 0x3fcaf924 0x00000000 0x3fcaf8d0
3fcaf8d0: 0xffffffff 0x3fcaf8d0 0x3fcaf8d0 0x00000000 0x3fcaf8e4 0xffffffff 0x3fcaf8e4 0x3fcaf8e4

ELF file SHA256: caf8d50e25f735ae

zhanzhaocheng, 2021-11-29:
@huexpub 从你的日志上分析，可能是由于调用 `mconfig_chain_master(……)` 传入了错误的参数引起的，将参考使用示例：[mconfig_example.c](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mconfig/main/mconfig_example.c#L96-L147)

huexpub, 2021-11-29:
> @huexpub 从你的日志上分析，可能是由于调用 `mconfig_chain_master(……)` 传入了错误的参数引起的，将参考使用示例：[mconfig_example.c](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mconfig/main/mconfig_example.c#L96-L147)

示例并不包括有线网络，代码是先检测eth链接，没有链接再启用wifi，过程没问题，不接eth也是没问题的，节点均可以收到配网数据，但只要接有线，无论是否启用wifi_sta链接到路由，均会造成上述问题，

closed 2021-12-01 by huexpub