espressif/esp-mdf #268: Support for esp8266

ankit-thealchemist, 2021-12-01:
Do we have the support for the esp8266 microcontrollers?

zhanzhaocheng, 2021-12-02:
Does not support esp8266, it is recommended that you use [esp32-c3](https://www.espressif.com/en/products/socs/esp32-c3), its cost is about the same as esp8266 and the performance is better

ankit-thealchemist, 2021-12-02:
Thanks for the support!

closed 2021-12-02 by ankit-thealchemist

mgonline, 2023-03-11:
> Does not support esp8266, it is recommended that you use [esp32-c3](https://www.espressif.com/en/products/socs/esp32-c3), its cost is about the same as esp8266 and the performance is better

Although the original question was answered clearly - _"this library does not support esp8266"_, I do not agree that pricing of esp8266 is same as that of esp32-c3. In fact, esp8266 is cheaper than the cheapest version of esp32 here in UK by at least 30-40%. The difference becomes apparent when you start purchasing ESPs in large quantities.

So, the question now is, is there any plan to support esp8266 in future? Are there any alternatives I can use meanwhile?

igrr, 2023-03-11:
@mgonline FWIW, taking Mouser for reference, ESP32-C3 is cheaper than the ESP8266 there:
* [ESP8266 chip](https://www.mouser.co.uk/ProductDetail/Espressif-Systems/ESP8266EX?qs=chTDxNqvsynorwcu2ADV3g%3D%3D), £1.32
* [ESP32-C3 chip](https://www.mouser.co.uk/ProductDetail/Espressif-Systems/ESP32-C3?qs=iLbezkQI%252BsiMXe7tMG%252Bo%2FA%3D%3D), £0.906