espressif/esp-mdf #269: failure CMake with idf.py

tatsutaigu, 2021-12-09:
## Environment

- Core (if using chip or module): ESP32, ESP32C3
- MDF version: cf50274 
- Operating System:		Ubuntu on WSL

## Problem Description
cmake cannot find project.cmake.

> $ idf.py menuconfig
> Executing action: menuconfig
> Running cmake in directory /home/tatsutaigu/esp/prj/mdf/master/esp32/router/build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/tatsutaigu/esp/prj/mdf/master/esp32/router"...
> CMake Error at CMakeLists.txt:8 (include):
>   include could not find load file:
> 
>     /home/tatsutaigu/esp/prj/mdf/master/esp32/router/../../../../project.cmake
> 
> 
> -- The C compiler identification is GNU 7.5.0
> -- The CXX compiler identification is GNU 7.5.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Configuring incomplete, errors occurred!
> See also "/home/tatsutaigu/esp/prj/mdf/master/esp32/router/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1
> 

### Code to reproduce this issue

root cause is CMakeLists.txt as below.
$ENV{MDF_PATH must remove $.
It can run idf.py menuconfig with modified CMakeLists.txt.

``` $ diff -u ~/esp/esp-mdf-master/examples/function_demo/mwifi/router/CMakeLists.txt ./CMakeLists.txt
--- /home/tatsutaigu/esp/esp-mdf-master/examples/function_demo/mwifi/router/CMakeLists.txt    2021-12-09 14:45:36.984505700 +0900
+++ ./CMakeLists.txt    2021-12-09 16:22:22.656280700 +0900
@@ -2,7 +2,7 @@
 # CMakeLists in this exact order for cmake to work correctly
 cmake_minimum_required(VERSION 3.5)

-if(NOT DEFINED $ENV{MDF_PATH})
+if(NOT DEFINED ENV{MDF_PATH})
     set(ENV{MDF_PATH} ${CMAKE_CURRENT_LIST_DIR}/../../../..)
 endif()
 include($ENV{MDF_PATH}/project.cmake)
``` 

I suggest to remove below code from CMakeLists.txt.

```
if(NOT DEFINED $ENV{MDF_PATH})
     set(ENV{MDF_PATH} ${CMAKE_CURRENT_LIST_DIR}/../../../..)
endif()
```

tonyshuang, 2021-12-09:
您好，信件已收到。

vitorrussi, 2021-12-09:
I made an PR fixing this issue almost 2 month ago here #253.

tatsutaigu, 2021-12-09:
OK, thank you.

closed 2021-12-09 by tatsutaigu