espressif/esp-mdf #271: mdf_mem buffer space of the memory record is full

XiotSamuel, 2021-12-18:
## Environment

- Development Kit:      M5 Atom Stamp
- Core (if using chip or module): ESP32-PICO-D4
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
   cf50274
- Development Env:		VisualCode
- Operating System:		[|MacOS]
- Power Supply:         USB

## Problem Description

I am using the light example as base for some redevelopment project. I found that the http server in the light example have some memory buffer issue when I send JSON message to the root node about 30 to 40 times continually. The API have report memory buffer issue, and the respond time for the http server have increase dramatic and same error message appear every time when I send JSON message to the root node.

### Expected Behavior

Expected the API can handle the message in large mount and able to clean up the buffer when it is full.

### Actual Behavior

The buffer is full but cannot clean up automatically. The buffer issue cannot solve until restart the IC.

### Steps to repropduce

1. Using postman to insert JSON message
2. mem_buffer error message will appear after around 30 times sending message to the root

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
<script src="https://gist.github.com/XiotSamuel/5571fa894f0a5fd7f73fdb9cd9c88747.js"></script>

## Debug Logs

I (133453) [mdf_mem, 129]: Memory record, num: 128, size: 13271
E (133461) [mdf_mem, 45]: The buffer space of the memory record is full
I (133462) [main, 325]: Node send, size: 2128, data: {"tid":"1","name":"light_7ad0","self_mac":"4c7525cc7ad0","parent_mac":"3c37867fb034","mesh_id":"3c37867fb034","version":"1.4.4-master","idf_version":"v4.3.1-dirty","mdf_version":"v1.0-156-gcf50274-dirty","mlink_version":2,"mlink_trigger":0,"rssi":-16,"layer":1,"tsf_time":"7327884739389","group":["010000000000","4c7525cc7ad0"],"characteristics":[{"cid":0,"name":"on","format":"int","perms":7,"value":1,"min":0,"max":3,"step":1},{"cid":1,"name":"hue","format":"int","perms":7,"value":360,"min":0,"max":360,"step":1},{"cid":2,"name":"saturation","format":"int","perms":7,"value":0,"min":0,"max":100,"step":1},{"cid":3,"name":"value","format":"int","perms":7,"value":100,"min":0,"max":100,"step":1},{"cid":4,"name":"color_temperature","format":"int","perms":7,"value":100,"min":0,"max":100,"step":1},{"cid":5,"name":"brightness","format":"int","perms":7,"value":100,"min":0,"max":100,"step":1},{"cid":6,"name":"mode","format":"int","perms":3,"value":3,"min":1,"max":3,"step":1},{"cid":7,"name":"lumen_level","format":"int","perms":3,"value":100,"min":0,"max":100,"step":1},{"cid":8,"name":"standby_lumen_level","format":"int","perms":3,"value":20,"min":0,"max":100,"step":1},{"cid":9,"name":"set_timer","format":"int","perms":3,"value":1,"min":0,"max":3,"step":1},{"cid":10,"name":"vancancy_timer","format":"int","perms":3,"value":1,"min":0,"max":120,"step":1},{"cid":11,"name":"fade_timer","format":"int","perms":3,"value":1,"min":0,"max":30,"step":1},{"cid":12,"name":"standby_timer","format":"int","perms":3,"value":1,"min":0,"max":240,"step":1},{"cid":13,"name":"CLO","format":"int","perms":3,"value":1,"min":0,"max":3,"step":1},{"cid":14,"name":"daylight_harvest","format":"int","perms":3,"value":1,"min":0,"max":3,"step":1},{"cid":15,"name":"level","format":"int","perms":3,"value":1,"min":0,"max":3,"step":1},{"cid":16,"name":"flash","format":"int","perms":3,"value":0,"min":0,"max":3,"step":1},{"cid":17,"name":"bh1750_value","format":"int","perms":1,"value":14,"min":0,"max":5000,"step":1},{"cid":18,"name":"ref_lux","format":"int","perms":1,"value":14,"min":0,"max":5000,"step":1}],"status_msg":"MDF_OK","status_code":0}
I (133463) [mdf_mem, 122]: (271) <iot_light: 392> ptr: 0x3ffbce30, size: 514
I (133659) [mdf_mem, 122]: (273) <iot_light: 399> ptr: 0x3ffc8cd4, size: 176
I (133670) [mdf_mem, 122]: (660) <mdebug_log: 191> ptr: 0x3ffd468c, size: 9
I (133681) [mdf_mem, 122]: (663) <mdebug_espnow: 110> ptr: 0x3ffd7280, size: 238
I (133682) [mdf_mem, 122]: (121039) <mdebug_espnow: 82> ptr: 0x3ffd7374, size: 239
I (133693) [mdf_mem, 122]: (667) <mlink_handle: 79> ptr: 0x3ffb865c, size: 108
I (133704) [mdf_mem, 122]: (668) <mlink_handle: 172> ptr: 0x3ffb86cc, size: 912
I (133705) [mdf_mem, 122]: (669) <mlink_trigger: 421> ptr: 0x3ffb8a60, size: 88
I (133717) [mdf_mem, 122]: (670) <mlink_espnow: 76> ptr: 0x3ffb0684, size: 1000
I (133728) [mdf_mem, 122]: (671) <mwifi: 301> ptr: 0x3ffb9828, size: 32
I (133729) [mdf_mem, 122]: (671) <mwifi: 306> ptr: 0x3ffb984c, size: 176
I (133740) [mdf_mem, 122]: (132419) <mlink_json: 178> ptr: 0x3ffef828, size: 2149
I (133751) [mdf_mem, 122]: (133461) <mwifi: 1373> ptr: 0x3ffeeb90, size: 584
I (133752) [mdf_mem, 122]: (10806) <mlink_httpd: 936> ptr: 0x3ffdff80, size: 80
I (133763) [mdf_mem, 122]: (10813) <mlink_notice: 169> ptr: 0x3ffeb654, size: 64
I (133774) [mdf_mem, 122]: (17029) <mlink_httpd: 346> ptr: 0x3ffdd6f0, size: 24
I (133786) [mdf_mem, 122]: (17030) <mlink_httpd: 410> ptr: 0x3ffec3b4, size: 477
I (133787) [mdf_mem, 122]: (17029) <mlink_httpd: 386> ptr: 0x3ffbb04c, size: 6
I (133798) [mdf_mem, 122]: (19163) <mlink_httpd: 346> ptr: 0x3ffb7d94, size: 24
I (133809) [mdf_mem, 122]: (19164) <mlink_httpd: 410> ptr: 0x3ffec598, size: 477
I (133810) [mdf_mem, 122]: (19164) <mlink_httpd: 386> ptr: 0x3ffbadc0, size: 6
I (133822) [mdf_mem, 122]: (20082) <mlink_httpd: 346> ptr: 0x3ffca0ec, size: 24
I (133833) [mdf_mem, 122]: (20083) <mlink_httpd: 410> ptr: 0x3ffe42ac, size: 477
I (133834) [mdf_mem, 122]: (20082) <mlink_httpd: 386> ptr: 0x3ffdd6c0, size: 6
I (133845) [mdf_mem, 122]: (20766) <mlink_httpd: 346> ptr: 0x3ffdd6d0, size: 24
I (133856) [mdf_mem, 122]: (20766) <mlink_httpd: 410> ptr: 0x3ffebb60, size: 477
I (133857) [mdf_mem, 122]: (20766) <mlink_httpd: 386> ptr: 0x3ffeb628, size: 6
I (133868) [mdf_mem, 122]: (21391) <mlink_httpd: 346> ptr: 0x3ffec1c8, size: 24
I (133879) [mdf_mem, 122]: (21392) <mlink_httpd: 410> ptr: 0x3ffebd44, size: 477
I (133880) [mdf_mem, 122]: (21392) <mlink_httpd: 386> ptr: 0x3ffec77c, size: 6
I (133892) [mdf_mem, 122]: (22134) <mlink_httpd: 346> ptr: 0x3ffe4700, size: 24
I (133903) [mdf_mem, 122]: (22134) <mlink_httpd: 410> ptr: 0x3ffebf28, size: 477
I (133905) [mdf_mem, 122]: (22134) <mlink_httpd: 386> ptr: 0x3ffe471c, size: 6
[... 78 lines cut ...]
I (134536) [mdf_mem, 122]: (69099) <mlink_httpd: 346> ptr: 0x3ffedd78, size: 24
I (134547) [mdf_mem, 122]: (69100) <mlink_httpd: 410> ptr: 0x3ffedda4, size: 36
I (134558) [mdf_mem, 122]: (69100) <mlink_httpd: 386> ptr: 0x3ffedd94, size: 6
I (134559) [mdf_mem, 122]: (70056) <mlink_httpd: 346> ptr: 0x3ffeddcc, size: 24
I (134571) [mdf_mem, 122]: (70057) <mlink_httpd: 410> ptr: 0x3ffed2b4, size: 36
I (134582) [mdf_mem, 122]: (70057) <mlink_httpd: 386> ptr: 0x3ffedde8, size: 6
I (134583) [mdf_mem, 122]: (71225) <mlink_httpd: 346> ptr: 0x3ffeddf8, size: 24
I (134594) [mdf_mem, 122]: (71226) <mlink_httpd: 410> ptr: 0x3ffed2ec, size: 36
I (134606) [mdf_mem, 122]: (71225) <mlink_httpd: 386> ptr: 0x3ffed2dc, size: 6
I (134607) [mdf_mem, 122]: (72457) <mlink_httpd: 346> ptr: 0x3ffed314, size: 24
I (134619) [mdf_mem, 122]: (72458) <mlink_httpd: 410> ptr: 0x3ffed330, size: 36
I (134620) [mdf_mem, 122]: (72458) <mlink_httpd: 386> ptr: 0x3ffeeb7c, size: 6
I (134631) [mdf_mem, 122]: (73325) <mlink_httpd: 346> ptr: 0x3ffed358, size: 24
I (134642) [mdf_mem, 122]: (73326) <mlink_httpd: 410> ptr: 0x3ffed384, size: 36
I (134654) [mdf_mem, 122]: (73325) <mlink_httpd: 386> ptr: 0x3ffed374, size: 6
I (134655) [mdf_mem, 122]: (105530) <mlink_httpd: 346> ptr: 0x3ffed3ac, size: 24
I (134666) [mdf_mem, 122]: (106552) <mlink_httpd: 410> ptr: 0x3ffed3d8, size: 36
I (134678) [mdf_mem, 129]: Memory record, num: 128, size: 12399
D (134681) [main, 89]: Root receive, addr: 4c:75:25:cc:7a:d0, size: 2128, data: {"tid":"1","name":"light_7ad0","self_mac":"4c7525cc7ad0","parent_mac":"3c37867fb034","mesh_id":"3c37867fb034","version":"1.4.4-master","idf_version":"v4.3.1-dirty","mdf_version":"v1.0-156-gcf50274-dirty","mlink_version":2,"mlink_trigger":0,"rssi":-16,"layer":1,"tsf_time":"7327884739389","group":["010000000000","4c7525cc7ad0"],"characteristics":[{"cid":0,"name":"on","format":"int","perms":7,"value":1,"min":0,"max":3,"step":1},{"cid":1,"name":"hue","format":"int","perms":7,"value":360,"min":0,"max":360,"step":1},{"cid":2,"name":"saturation","format":"int","perms":7,"value":0,"min":0,"max":100,"step":1},{"cid":3,"name":"value","format":"int","perms":7,"value":100,"min":0,"max":100,"step":1},{"cid":4," [... cut ...] :1},{"cid":17,"name":"bh1750_value","format":"int","perms":1,"value":14,"min":0,"max":5000,"step":1},{"cid":18,"name":"ref_lux","format":"int","perms":1,"value":14,"min":0,"max":5000,"step":1}],"status_msg":"MDF_OK","status_code":0}

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

XiotSamuel, 2021-12-18:
I just find out that the mem_buf become normal again after I disable all mqtt related function, and reason for that???

zhanzhaocheng, 2021-12-20:
This is because you failed to release the memory of http_data after calling `mesh_mqtt_read()`, so a memory leak occurred

[Here](https://gist.github.com/XiotSamuel/5571fa894f0a5fd7f73fdb9cd9c88747#file-main-c-L159-L175) amended as follows:
```c
         mesh_mqtt_data_t *request = NULL;
         ret = mesh_mqtt_read(&request, pdMS_TO_TICKS(500));

         if (ret != MDF_OK) {
             // continue;
             goto FREE_MEM;
         }
```

XiotSamuel, 2021-12-20:
Yep, you are correct.

I have also test

        MDF_ERROR_GOTO(ret != MDF_OK, FREE_MEM, "<%s> mesh_mqtt_read", mdf_err_to_name(ret));

and

        if (ret != MDF_OK) {
            //continue;
            goto FREE_MEM;
        }

both method also work. Many thanks.

closed 2021-12-20 by XiotSamuel