espressif/esp-mdf #275: esp-mdf安装错误

llltong, 2021-12-31:
不清楚为什么使用./install.sh安装后，会弹出这个，请问怎么解决？

root@l-VirtualBox:/usr/esp32/esp-mdf/esp-idf# ./install.sh
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.6.9
"python3" has been detected
Installing ESP-IDF tools
Selected targets are: esp32, esp32s2, esp32c3, esp32s3
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20211111 (already installed)
Installing Python environment and packages
Creating a new Python environment in /root/.espressif/python_env/idf4.3_py3.6_env
Usage: virtualenv.py [OPTIONS] DEST_DIR

virtualenv.py: error: no such option: --seeder
Traceback (most recent call last):
  File "/usr/esp32/esp-mdf/esp-idf/tools/idf_tools.py", line 1822, in <module>
    main(sys.argv[1:])
  File "/usr/esp32/esp-mdf/esp-idf/tools/idf_tools.py", line 1818, in main
    action_func(args)
  File "/usr/esp32/esp-mdf/esp-idf/tools/idf_tools.py", line 1519, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usrb/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'virtualenv', '--seeder', 'pip', '/root/.espressif/python_env/idf4.3_py3.6_env']' returned non-zero exit status 2.
root@l-VirtualBox:/usr/esp32/esp-mdf/esp-idf#

Jiangyafeng, 2022-01-10:
@llltong 你可以使用这个 /usr/bin/python3', '-m', 'virtualenv', '--seeder', 'pip', '/root/.espressif/python_env/idf4.3_py3.6_env' 指令安装一下，看是否可以解决？这个应该是你的这个版本缺少了组件导致的。

llltong, 2022-01-17:
> 

这是什么命令？没看懂。。。。命令格式看着奇怪

Jiangyafeng, 2022-02-12:
@llltong sudo apt-get install python-virtualenv，安装一下，看是否可以解决你的问题？最好是你的linux系统默认python要是v3.x 版本，你直接使用python --version查询一下，如果不是，请将其切换为python3版本。