espressif/esp-mdf #277: English version of alibabacloud compatable with espmesh ?

ThatBigPrint, 2022-01-21:
hi guys, 

im trying to set up esp mesh aliyun and all instructions are in Chinese which is ok, but the international version of the website looks and has different nomenclature so can someone guide me, please? 

the wording on the international version is called "IoT Platform" I presume that is the equivalent to "Life IoT Platform" ?

can you please guide me or translate as the instructions on link below I cant translate the pictures 
https://github.com/espressif/esp-aliyun

I presume the link above is relevant to esp mesh?

as the link below only covers inputting the key and picture but in my version (international) I cant find out how to get the picture? as described in the link below?

https://github.com/EspressifApp/EspMeshForiOS

also as a manufacturer how would this work for our customers? would they need to create an Ali account or would we induct them on our end?

zhanzhaocheng, 2022-01-24:
Since esp mesh aliyun is aimed at Chinese customers, there is no English version

To simplify the development of Wi-Fi Mesh connected cloud services, we are currently developing a test [ip for
  Mesh](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) and [esp-gateway](https://github.com/espressif/esp-gateway) solutions, they can be easily adapted to any cloud platform, we will also provide an example of Wi-Fi Mesh connecting ESP-RainMaker, I suggest you wait for a while, and we will update it in time if there is progress you
If your project is very urgent, you can contact one of our business: https://www.espressif.com/en/contact-us/technical-inquiries

ThatBigPrint, 2022-01-24:
so the platform i need to get my app developer to work with is going to be rainmaker? is there a way to get esp mesh to work with rainmaker without a gateway at the moment or will it always need a gateway? 

and how long will this development take roughly months?

zhanzhaocheng, 2022-01-24:
This solution does not require a gateway and can quickly switch between different cloud platforms, [ESP-RainMaker](https://rainmaker.espressif.com/) is one of the examples we provide

The function has been implemented, and we are currently optimizing the code internally. It is expected to be in early March 2022.

ThatBigPrint, 2022-01-24:
will this be a simple OTA update if we used esp mesh for the interim, then upgraded later on?

zhanzhaocheng, 2022-01-24:
The difference between the two is still relatively big from the code, but it can be upgraded through OTA,
It is recommended that you learn about ESP-RainMaker first

zhanzhaocheng, 2022-01-25:
What we're doing now is making RainMaker compatible with ESP-MESH, But we are currently testing based on 50 devices. Does your network have requirements for communication rate, networking time, and delay?

ThatBigPrint, 2022-01-25:
our requirements are similar to esp mesh but not that critical, mainly easy provisioning and low as possible latency...

ThatBigPrint, 2022-01-25:
if you can share the source for this function I can test for our used case ?

zhanzhaocheng, 2022-01-25:
At present, we are in the process of code review, which will not be available until mid February at the earliest

ThatBigPrint, 2022-01-25:
ok, so that was the answer i was after.... so instead of investing in an app based on the old esp mesh one I may as well start with the rainmaker? 

three questions 

1. can rainmaker provision like esp mesh where it just prompts you to add a nearby device? 
2. do we use esp mesh firmware like we were before ?
3. when I have a customer do they need to create a rainmaker account or do they subscribe to the retailer's app? i would prefer them purchase through ourselves

xcguang, 2022-01-26:
@ThatBigPrint  Current mesh feature of [esp-gateway](https://github.com/espressif/esp-gateway) is a simple demo, you can try it by enable `SET_VENDOR_IE`, but we will next refactor it in the next month.