espressif/esp-mdf #278: Is ESP-MDF being discontinued?

vitorrussi, 2022-02-07:
The last commit was in late October, which were said that v2.0 would be released and until now we don't have any news, the 30-month support on ESP-IDF 3.3.x is ending this month and we don't know when the new release is going to be on. This framework will be updated? If won't, then we can start migrating to ESP-IDF's mesh framework or any other mesh framework, but if the development continues then just some news on the development status and an expected date will be greatly helpful.

moefear85, 2022-02-09:
same question.

eos1d3, 2022-02-09:
The real question is not just new release but the extremely low activities on MDF. If major bugs are found, it is not likely fixes will be available in short time. 

This is one of the nice libraries for ESP. But using it for real product is very dangerous at there is virtually no support at all.

XiotSamuel, 2022-02-14:
I have one project need to implement a firmware with mdf. good luck to me. T.T

jci-zimm, 2022-02-14:
@XiotSamuel , i'm not really sure the MDF is strictly needed for mesh all together. The MDF just represents wrapping components the IDF. THB, i'd rather not be tied to a particular MDF version because that may lag IDF. You can implement FW update relatively easily given the IDF component for firmware update. In that case though, you would be managing the FW update across the mesh yourself.

XiotSamuel, 2022-02-15:
@jci-zimm I believe mdf is one of the method to implement the mesh. IDF have Bluetooth mesh components provided. mdf is not the only way to implement mesh network. On the other hand, the mdf is alway tie to one particular version of idf. May be this design make the mdf subject to not receive updated idf component? Not sure why have this design. Anyway, my project have successfully implement a 100+ node mesh network and mlink in the network, but we found that it is quite hard to add the component like mqtt, security for the mesh network. The intermessaging with mini write also display memory leak issue and cause the root node to restart. Hope they can fix it in next version. 🙏

Jiangyafeng, 2022-02-17:
Hi All,

Sorry to cause you trouble. The maintenance of this warehouse has been delayed due to the transfer of internal staff. Maintenance and updates will continue in the future.

eos1d3, 2022-02-17:
Really very interesting for using MDF in our products. If Espressif can allocate more resources will be a lot better. I am seeing a lot issues are pending and hope Espressif can address them.

tatsutaigu, 2022-04-12:
@Jiangyafeng,
I understand your resource is not enough.
And, I worry the future of ESP-MDF because it is pending from 2021 October.
I have been waiting for it for a half of year.
I would like to know ESP-MDF release plan.
Please provide us with information that we can feel comfortable with.

Jiangyafeng, 2022-04-19:
Hi @tatsutaigu
Thank you very much for your attention. It is true that this database has not been updated in time for some time. We will release a version as soon as possible after the end of COVID-19.

habablaxe, 2022-06-02:
> Hi @tatsutaigu Thank you very much for your attention. It is true that this database has not been updated in time for some time. We will release a version as soon as possible after the end of COVID-19.

any news?

vitorrussi, 2022-06-30:
Hey @Jiangyafeng
Would you give us an update about the current development status? When the new version will be released, or when maintenance and updates will resume

PatrickV82, 2022-07-14:
are there any updates?

willianrosner, 2022-08-01:
Hello,

We still don't have any concrete position on the new updates, I have 700 devices that work in mesh network using mdf, but I'm starting not to link the new equipment through the MDF, but only through the IDF, because I'm taking too much risk without being sure that we will have improvements here. I really like developing products with Espressif, but in terms of MDF we are looking at ships. Any news?

tatsutaigu, 2022-08-02:
@Jiangyafeng,
I heard that the lockdown COVID-19 in Shanghai ended at the end of June.
However, I feel Espressif have not start maintenance  ESP-MDF yet.
The fact that the official repository has been suspended for 10 months is an anomaly.

Espressif's first priority is Matter support?
I understand Matter is game changer at home network.
But, we are hard to use Matter because we are hard to get DAC, PAI and any certification and operate any certification servers.

ESP-MDF is our best solution.
When will start maintenance and updating ESP-MDF?
Would you please disclose ESP-MDF release plan for us.

mmrein, 2022-08-02:
Btw you can also add MDF as local components to your IDF project - that gives you more variability while you can edit MDF directly. This way you can also use MDF with Eclipse Plugin. I ultimately tried it with IDF v5.0 and it does somehow work (not without bugs, but mostly it does work).

GaelicThunder, 2022-08-10:
Hi @mmrein , mind to share how you got it work with the 5? Im trying so hard to make it build you don't even know

mmrein, 2022-08-11:
@GaelicThunder Well, exact steps would depend on your project and IDE you use (if any), but in general you first have to  convert your MDF project to regular IDF project with MDF components before these steps or it will get too messy. That **could** be as easy as:
- copy MDF components to your project folder's components
- comment out MDF_PATH line in your `CMakeLists.txt`
- setting `MDF_VER` string somewhere (I put it simply to `mdf_common.h` in my project components as `#define MDF_VER        "v2.0b"`). 
- then make full clean (probably also delete build folder and rename `sdkconfig` file to `sdkconfig.defaults`)
- then try rebuilding your project (in environment where MDF_PATH variable is not set).

If that went well and you can build project in that way, make full clean of your project again, now including:
- delete `build` folder
- rename your generated `sdkconfig` file to `sdkconfig.defaults`
- as I used Eclipse with legacy make in my project originally, I also had to delete hidden files and folder: `.cproject, .project, .settings/`, then I could import as existing IDF project into new Eclipse IDE with Espressif Plugin (configure the IDE separatelly in the first place and test some hello world project)
- In your project's root `CMakeLists.txt` change to `cmake_minimum_required(VERSION 3.16)`

Then you can try building your project using CMake build system, which will end up in lot of errors and warnings. At this point the *migration guide* will be most helpfull: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/index.html. 

Specifically this chapter solved most of errors: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/build-system.html#specify-component-requirements-explicitly as I simply had to add some component names to **REQURES** in certain CMakeLists.txt as they are not included by default. For example this error:

`../components/mcommon/include/mdf_common.h:48:10: fatal error: esp_event.h: No such file or directory`
-> Solved by adding `esp_event` to **REQURES** in `./components/mcommon/CMakeLists.txt`

What you will most likely see with MDF is `unknown type name 'xQueueHandle'`. Easiest is to enable the `CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY` option and add `freertos` component name to REQURES in project main's CMakeLists.txt.

When you can face `implicit declaration of function ...` errors, check the document where you will probably find that original function was renamed, deprecated and replaced with another one or even moved to different location. That depends on specific functions you use.

This way I was able to connect root and node, seemingly working as before. One interesting bug is that if I restart node when it is connected to root, the root won't re-connect it. If I let it disconnect first (just let the node off for like 30s), then it will reconnect OK, so there is probably something wrong with routing table functionality. I did not dig into that at all.

GaelicThunder, 2022-08-11:
@mmrein thanks for the super detailed answer! Gonna try this days 😁👍

habablaxe, 2022-09-14:
Hi @Jiangyafeng
Any news about MDF release or development status? please tell us any information

PatrickV82, 2023-03-06:
Hi @Jiangyafeng 
Any news about MDF development status?

tonyshuang, 2023-03-06:
您好，信件已收到。

mmrein, 2023-03-18:
I just realized that if MDF followed same support policy of 30 months as IDF, then it ended on 21.10.2022. RIP

habablaxe, 2023-03-20:
> IDFv-5 wont compile
> I just realized that if MDF followed same support policy of 30 months as IDF, then it ended on 21.10.2022. RIP
but, it’s also strange that they announce MDF at the exhibition https://www.youtube.com/watch?v=s0Z4FWi1Llw , although they abandoned it for more than a year

mmrein, 2023-03-20:
I mean, did they really or did it just fit the picture better that ESP-WIFI-MESH...

tonyshuang, 2023-06-23:
您好，信件已收到。