espressif/esp-mdf #281: Suggested parenthesis around expression in mwifi.c

mmrein, 2022-03-24:
## Environment

- Possibly related to all - bug in mwifi.c source file
- MDF version: cf502740f5a6c82a0dc3059e7591c262795f70b2 (current master)

## Problem Description

Missing parenthesis around expression: https://github.com/espressif/esp-mdf/blob/cf502740f5a6c82a0dc3059e7591c262795f70b2/components/mwifi/mwifi.c#L704

I did not really experience any issue while running my app that should be related, but above expression without parenthesis does not seem right.

### Expected Behavior

Build without warning, explicit use of parenthesis.

### Actual Behavior

Builds with warning, possible unexpected behaviour.

Jiangyafeng, 2022-04-19:
Hi @mmrein 
There is nothing wrong with this, you can look at the link:
https://github.com/espressif/esp-mdf/blob/master/components/mcommon/include/mdf_err.h#L115, 
the print here, ‘err’ print is the reason for the error.

Did you compile any warning? Could you please provide what kind of warning it is?

mmrein, 2022-04-20:
Hi @Jiangyafeng, I see nothing wrong with MDF_ERROR_CHECK macro statement. 

Warning literally says (in Eclipse):
```
Suggested parenthesis around expression 'MDF_ERROR_CHECK(ret != ESP_OK && !(flag & MESH_DATA_GROUP && ret == ESP_ERR_MESH_DISCARD), ret,
                        "Node failed to send packets, dest_addr: " MACSTR
                        ", flag: 0x%02x, opt->type: 0x%02x, opt->len: %d, data->tos: %d, data: %p, size: %d",
                        MAC2STR(dest_addr->addr), flag, opt->type, opt->len, mesh_data.tos, mesh_data.data, mesh_data.size)'
mwifi.c		 /components/mwifi	line 704	Code Analysis Problem
```

Jiangyafeng, 2022-04-21:
Well, thank you very much for the offer. You made the code clearer. This will be updated later.