espressif/esp-mdf #283: mconfig_chain_master永久堵塞

fzboffice, 2022-03-31:
某些情况下(猜测ram不足),mconfig_chain_master执行后没有触发start事件 当再次调用mconfig_chain_master时会被锁锁住

Jiangyafeng, 2022-04-14:
@fzboffice 方便说明一下复现的流程吗？怎么产生的再次调用 mconfig_chain_master 时会被锁锁住。

fzboffice, 2022-04-19:
@Jiangyafeng 我并不能够实现复现 是偶发的 但是我观察到发生这个事情的时候不会有mconfig start event上报给我 并且mconfig_chain_master能够返回不会阻塞 并且下一次调用mconfig_chain_master百分百卡住

Jiangyafeng, 2022-04-20:
能否提供一个比较容易复现的 demo 或者你复现时候的 log 发送一下。更希望你能提供一下比较容易复现的大概环境和demo流程，我这边可以基于你的压测一下，复现的话比较容易找到问题所在。

fzboffice, 2022-04-21:
@Jiangyafeng 已经能够实现百分百复现了，就是在调用mconfig_chain_master之前故意把MALLOC_CAP_INTERNAL申请到极低的水平，当此mconfig_chain_master能够正常返回，但是下次调用mconfig_chain_master就永久阻塞了

Jiangyafeng, 2022-04-22:
@fzboffice 你的意思是 ram 剩余内存极低的情况下，发生的吗？能否提供一下设备端的log？

fzboffice, 2022-04-22:
@Jiangyafeng 我打开了MDF的log级别为debug 然后申请大量的ram（故意行为），然后调用mespnow_init打印出I (45613) ESPNOW: espnow [version: 1.0] init 然后调用mconfig_chain_master 无任何打印 正常return 此时再次调用mconfig_chain_master 无任何打印

Jiangyafeng, 2022-04-25:
@fzboffice 你那边有复现的demo吗？我可以具体看一下什么原因。我害怕你将其剩余内存没有了，导致程序crash了吧？还有你的剩余内存还有多少？发送一下你的设备端log？

内存剩余过少是不行的，因为wifi在运行的时候是需要留下一定余量的动态内存的。

fzboffice, 2022-04-25:
[demo.zip](https://github.com/espressif/esp-mdf/files/8552200/demo.zip)

Jiangyafeng, 2022-04-25:
@fzboffice 非常感谢你的反馈，确实存在这个问题。这里一直被阻塞在那里是因为信号量一直没有获取到，而且设置的是 portMAX_DELAY。
阻塞位置：[portMAX_DELAY](https://github.com/espressif/esp-mdf/blob/master/components/mconfig/mconfig_chain.c#L598)
你可以在 https://github.com/espressif/esp-mdf/blob/master/components/mconfig/mconfig_chain.c#L588 下，添加如下修改：
`    
if (!g_chain_master_exit_sem) {
        g_chain_master_exit_sem = xSemaphoreCreateBinary();
 }
`
将其 https://github.com/espressif/esp-mdf/blob/master/components/mconfig/mconfig_chain.c#L595，这个删除掉就可以了。

fzboffice, 2022-04-26:
@Jiangyafeng 感谢解答阻塞问题，另外我认为如果mconfig_chain_master()不能正常启动并激发start event 则应该在event中单独加一个失败event或直接由mconfig_chain_master()返回失败

Jiangyafeng, 2022-04-26:
对于 mconfig_chain_master_task 无法运行的原因是因为剩余内存太少了，从而导致 xTaskCreate 无法创建成功导致的，位置：https://github.com/espressif/esp-mdf/blob/master/components/mconfig/mconfig_chain.c#L613，

这个你可以通过 uxTaskGetStackHighWaterMark 来具体看到每个 task 的剩余可用情况。