espressif/esp-mdf #288: 使用官方例程下的simple_ota_example.c一直显示头文件protocol_examples_common.h找不到

amarelo-t, 2022-05-06:
你好，我在使用官方实例下的simple_ota_example.c一直显示头文件protocol_examples_common.h找不到，请问这个问题该如何解决呢
[image]

amarelo-t, 2022-05-06:
你好，我去monitor的时候显示下面错误的原因，我想请问这个怎么办呢，谢谢！
[image]

335980889, 2022-05-21:
1、连接WIFI了吗
2、http服务器开了吗
3、esp32的ota地址对吗

kid-king-x, 2024-07-15:
我看了看官方给的那个例程，是要在idf_component.yml文件里面加入  protocol_examples_common:
    path: ${IDF_PATH}/examples/common_components/protocol_examples_common这个的，像我下面那个一样（不过我下面那个还加了摄像头需要的）
[image]
[image]
这个一个是官方的，一个是我自己的

tonyshuang, 2024-07-15:
您好，信件已收到。