espressif/esp-mdf #290: [求助]关于win10环境下，ESP-MDF运行install.bat出现python的错误

335980889, 2022-05-21:
python版本 : 3.7.8

Setting up Python environment
Traceback (most recent call last):
  File "C:\Users\Administrator\Desktop\esp-mdf-v1.0\esp-idf\tools\idf_tools.py", line 1358, in <module>
    main(sys.argv[1:])
  File "C:\Users\Administrator\Desktop\esp-mdf-v1.0\esp-idf\tools\idf_tools.py", line 1354, in main
    action_func(args)
  File "C:\Users\Administrator\Desktop\esp-mdf-v1.0\esp-idf\tools\idf_tools.py", line 1144, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "C:\Users\Administrator\Desktop\esp-mdf-v1.0\esp-idf\tools\idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 488, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] 系统找不到指定的文件。

tonyshuang, 2022-05-21:
您好，信件已收到。

335980889, 2022-05-21:
我用vmware装个了win10，重新安装了一遍就可以了，是物理机环境问题。

closed 2022-05-21 by 335980889