espressif/esp-mdf #292: vscode  下使用esp-mdf新建工程编译出错

yeqinggui, 2022-05-25:
通过vscode  安装esp-mdf，从mdf导入例程，新建一个项目，编译报错，有遇到类似问题吗?
> Executing task: ninja  <

ninja: error: loading 'build.ninja': 系统找不到指定的文件。

终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。

Jiangyafeng, 2022-06-30:
Hi @yeqinggui，你可以先将 idf 环境搭建好，然后能够运行，因为 MDF 也是依托于idf环境的。你可以参看连接：https://github.com/espressif/vscode-esp-idf-extension

需要在 vscode 配置设置中将 idf.espMdfPath 或 idf.espMdfPathWin（Windows中）设置为esp-mdf目录路径

有一个扩展命令可以克隆 ESP-MDF 并自动设置此设置，称为 ESP-IDF：install ESP-MDF

具体定义的连接：https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md#other-frameworks-specific-settings

sxc1143106070, 2024-04-25:
> Hi @yeqinggui，你可以先将 idf 环境搭建好，然后能够运行，因为 MDF 也是依托于idf环境的。你可以参看连接：https://github.com/espressif/vscode-esp-idf-extension
>
> 需要在 vscode 配置设置中将 idf.espMdfPath 或 idf.espMdfPathWin（Windows中）设置为esp-mdf目录路径 [...]

你好，我现在IDF的环境没有任何问题，然后通过 ESP-IDF：install ESP-MDF 命令安装了MDF环境，但是编译例程时老是报错，能麻烦你指点下吗

iwonako, 2024-11-21:
> 你好，我现在IDF的环境没有任何问题，然后通过 ESP-IDF：install ESP-MDF 命令安装了MDF环境，但是编译例程时老是报错，能麻烦你指点下吗

请问解决了吗