espressif/esp-mdf #297: 允许信道切换但断线后不能进行全信道扫描

fzboffice, 2022-07-07:
channel_switch_disable == false 的情况下 只会在开机find network，但是当和mesh断开 并达到 attemp中定义的60次之后 不会进行一个全信道扫描

Jiangyafeng, 2022-07-13:
Hi @fzboffice  请在 event_loop_cb 事件里面添加如下代码：
`        case MDF_EVENT_MWIFI_PARENT_CONNECTED:
            MDF_LOGI("Parent is connected on station interface");

            if (esp_mesh_is_root()) {
                esp_netif_dhcpc_stop(netif_sta);
                esp_netif_dhcpc_start(netif_sta);
            }

            break;
`
请测试一下如何？

fzboffice, 2022-07-13:
@Jiangyafeng 你好，我指的是node节点与mesh断开后 比如root掉电重新上电然后root切换到了一个新的信道，此时这个node节点在断连后无法进行一个全信道扫描，只在上次断开的信道上一直扫描

Jiangyafeng, 2022-07-19:
@fzboffice 方便描述一下你是怎么复现的吗？我这边基于你说的情况，在router例程下测试，是没有复现的，他是能正常切换的。
正常情况下 root节点切换信道的话，node节点会有如下时间发生的：
_W (1159144) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:72, no_wnd_count:0, t
imeout_count:7
I (1159481) wifi:switch to channel 2
I (1159481) wifi:ap channel adjust o:8,2 n:2,1
I (1159481) wifi:new:<2,0>, old:<8,2>, ap:<2,1>, sta:<2,0>, prof:8
I (1159788) wifi:new:<2,1>, old:<2,0>, ap:<2,1>, sta:<2,1>, prof:2
W (1160345) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:72, no_wnd_count:0, t
imeout_count:8
I (1161488) [router_example, 269]: System information, channel: 2, layer: 2, self mac: 7c:87:ce:e3:56:48, parent b
ssid: b4:e6:2d:eb:03:96, parent rssi: -21, node num: 2, free heap: 178080_

是否可以提供一下复现简单的demo，还有描述一下复现场景。

fzboffice, 2022-07-19:
@Jiangyafeng 可能是我表达不对，demo暂时有点困难，描述一下场景，场景是这样的，fix root，root和node刚开始是配置好连接着的，此时是信道x，然后root关机，root开机后启动mesh前进行一个信道切换到y，没有调用esp_mesh_switch_channel，然后启动mesh后，node无法连接到root，根本原因在于node在断开事件达到 attempt_count定义的60次后并没有进行full channel scan，而是永远在信道x上扫描

mmrein, 2022-07-19:
I'm having the same issue (channel switching enabled but full channel scan is not performed after disconnection).