espressif/esp-mdf #298: esp-mdf是否支持广播模式

tsk15535904190, 2022-07-19:
开发者你好：
esp-mdf作为esp-now协议的应用是否支持广播模式，官网最新的esp-now手册显示为2016年，不支持广播模式，六年过去了，如今esp-now协议是否支持广播模式？

Jiangyafeng, 2022-07-20:
@tsk15535904190  在 esp-now 的应用库里面已经支持广播模式，你可以参看连接：https://github.com/espressif/esp-now，
关于广播模式你可以查看 espnow.c 驱动实现，连接：https://github.com/espressif/esp-now/blob/master/components/espnow/src/espnow.c#L120

tsk15535904190, 2022-07-23:
感谢您的回复！esp-now协议在广播时，是否支持路由转发，我在源代码中看到了类似的踪迹

Jiangyafeng, 2022-08-01:
@tsk15535904190 首先esp-now是私有协议，详细的你可以查看链接：https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html，这里是有相关的说明的。

ESP 模组都是支持 wifi 和 now 的共存，可以将其接收到 now 的数据，通过wifi转发出去给路由器，路由在转发到外网的机制。
更多相关的 now 开发问题，可以通过 https://github.com/espressif/esp-now/issues 来询问。