espressif/esp-mdf #303: ESP32 doesn't boot with mupgrade fallback version flag with two cores after IDF 4.4.2 update (IDFGH-7996)

VitorBFreitas, 2022-08-08:
## Environment

- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.4.2
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows
- Environment type: PowerShell
- Using an IDE?: Yes (VScode espressif extension)
- Power Supply: USB

## Problem Description
ESP32 doesn't complete boot with mupgrade fallback version flag with two cores enabled after update from idf 4.4.1 to 4.4.2

### Expected Behavior
ESP32 boot normally

### Actual Behavior
ESP32 freeze while booting

### Steps to reproduce
1. create a project with mupgrade example
2. open menuconfig
3. disable "Run FreeRTOS only on first core" flag
4. enable "Fall back to the previous version" flag
6. Build and Flash

### Code to reproduce this issue
Mupgrade example at esp-mdf\examples\function_demo\mupgrade

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6664
load:0x40078000,len:14848
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (27) boot: compile time 13:25:51
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (75) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (83) boot:  2 app0             OTA app          00 10 00010000 001b0000
I (90) boot:  3 app1             OTA app          00 11 001c0000 001b0000
I (98) boot:  4 spiffs           Unknown data     01 82 00370000 00060000
I (105) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (113) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (120) boot: End of partition table
I (124) boot: No factory image, trying OTA 0
I (129) boot_comm: chip revision: 3, min. application chip revision: 0
I (137) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=38a60h (232032) map
I (229) esp_image: segment 1: paddr=00048a88 vaddr=3ffb0000 size=03c60h ( 15456) load
I (235) esp_image: segment 2: paddr=0004c6f0 vaddr=40080000 size=03928h ( 14632) load
I (242) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=dd4d0h (906448) map
I (571) esp_image: segment 4: paddr=0012d4f8 vaddr=40083928 size=11bd4h ( 72660) load
[... 85 lines cut ...]

D (1061) cpu_start: calling init function: 0x40103694
0x40103694: s_set_default_wifi_log_level at C:/esp/esp-idf/components/esp_wifi/src/wifi_init.c:63

D (1066) cpu_start: calling init function: 0x400f7704
0x400f7704: esp_ipc_init at C:/esp/esp-idf/components/esp_ipc/src/esp_ipc.c:105

D (1072) cpu_start: calling init function: 0x400ea0b0
0x400ea0b0: mupgrade_partition_switch at C:/esp/esp-mdf/components/mupgrade/mupgrade_check.c:134

D (1076) [mupgrade_check, 138]: Add an identifier to the firmware: ** MUPGRADE_FIRMWARE_FLAG **
D (1085) cpu_start: calling init function: 0x400d4164
0x400d4164: esp_reset_reason_init at C:/esp/esp-idf/components/esp_system/port/soc/esp32/reset_reason.c:68

D (1090) cpu_start: calling init function: 0x400d355c
0x400d355c: esp_ota_init_app_elf_sha256 at C:/esp/esp-idf/components/app_update/esp_app_desc.c:68

D (1095) intr_alloc: Connected src 17 to int 3 (cpu 0)
D (1101) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (1106) cpu_start: Starting scheduler on PRO CPU.
D (1112)D (1112) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (1117) cpu_start: Starting scheduler on APP CPU.
 partition: Loading the partition table
D (1127) partition: Partition table MD5 verified
```

Jiangyafeng, 2022-08-08:
@VitorBFreitas Thank you very much for this question. It is true that the latest idf v4.4.2 has this problem in adapting to mdf, and this problem will be fixed as soon as possible. If there is any new progress, it will be updated as soon as possible.

Jiangyafeng, 2022-08-11:
@VitorBFreitas You can not enable Component config -> MDF Mupgrade -> Fall back to the previous version. There is no problem.

VitorBFreitas, 2022-08-11:
@Jiangyafeng Unfortunetly this is a feature that I'm using in my actual project to prevent loose communication with MQTT broker case a bad firmware is flashed.

mmrein, 2022-12-21:
Is there some fix for this? Does anyone know what causes this? Does v4.4.3 have same problem? 

I'm planning to make some changes in my project which would require some safety measures with possibility to revert to previous firmware in case wrong image has been flashed.

VitorBFreitas, 2022-12-21:
@mmrein It's occurs with esp-idf v4.4.3 as well.

mmrein, 2023-03-10:
I didn't realize this occurs even after first flash, not just after OTA. So I can confirm this bug is still present in v4.4.4

mmrein, 2023-03-19:
Found some kind of clue - with debul log level set to Verbose the app seems to be starting, but on level Debug or less it freezes:

```
--- LOG OUTPUT DEFAULT VERBOSITY: DEBUG
D (1564) intr_alloc: Connected src 24 to int 3 (cpu 0) 
I (1569) cpu_start: Starting scheduler on PRO CPU. 
D (1574) intr_alloc: Connected src 25 to int 2 (cpu 1) 
I (1579) cpu_start: Starting scheduler on APP CPU. 
D (1574) partition: Loading the partition table 
D (1594) partition: Partition table MD5 verified
--- FREEZE
```

```
--- LOG OUTPUT DEFAULT VERBOSITY: VERBOSE
D (2099) intr_alloc: Connected src 24 to int 3 (cpu 0) 
I (2104) cpu_start: Starting scheduler on PRO CPU. 
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args 
V (2116) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E 
D (2124) intr_alloc: Connected src 25 to int 2 (cpu 1) 
I (2129) cpu_start: Starting scheduler on APP CPU. 
D (2110) partition: Loading the partition table 
V (2145) calculated md5: 0x3ffb7cb8   86 8a 0e ec e7 f1 b4 8a  6f 58 5b ed dd 1e ed e3  |........oX[.....| 
V (2149) stored md5: 0x3f45e190   86 8a 0e ec e7 f1 b4 8a  6f 58 5b ed dd 1e ed e3  |........oX[.....| 
D (2158) partition: Partition table MD5 verified 
D (2167) nvs: nvs_open_from_partition ESP-MDF 1 
D (2168) nvs: nvs_set_blob mupgrade_count 4 
D (2172) nvs: nvs_close 1 
D (2174) [mupgrade_check, 127]: version_fallback_task exit 
D (2180) heap_init: New heap initialised at 0x3ffe0440 
D (2185) heap_init: New heap initialised at 0x3ffe4350 
I (2190) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations 
D (2199) spiram: Allocating block of size 32768 bytes 
V (2204) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args 
V (2204) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE 
D (2204) intr_alloc: Connected src 16 to int 9 (cpu 0) 
--- GOING...
```

VitorBFreitas, 2023-03-20:
Thanks for share your tests! I'm sure this will help those who get stuck in it.