espressif/esp-mdf #305: Heap memory management bug (IDFGH-8154)

stbnps, 2022-08-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.1 and 4.4.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

Multiple ESP32 models (WROOM/WROVER)

### Power Supply used.

USB

### What is the expected behavior?

Perform IDF functions with no heap memory management errors.

### What is the actual behavior?

I have multiple stack traces, most of them related to __mwifi_root_read. All of them were generated with the same setup (code/hardware), using a mesh of 3 ESP32 devices. Even if __mwifi_root_read is not in the stack trace, all the errors were generated in the root node.

Stack traces indicate that this may be a memory management problem. The errors seem related to free block management.

Trace 1:

```
assert failed: block_merge_prev heap_tlsf.c:350 (block_is_free(prev) && "prev block is not free though marked as such")

Backtrace:0x40081d32:0x3ffe2d700x40089649:0x3ffe2d90 0x4008fe81:0x3ffe2db0 0x4008f1d9:0x3ffe2ed0 0x4008faaa:0x3ffe2ef0 0x40081fa2:0x3ffe2f10 0x4008feb1:0x3ffe2f30 0x40168c4f:0x3ffe2f50 0x40168cf6:0x3ffe2f70 0x40145a73:0x3ffe2fa0 0x40146e67:0x3ffe2ff0 0x400f8593:0x3ffe3020 0x400e6ed3:0x3ffe30c0 0x4008cdad:0x3ffe3210 
0x40081d32: panic_abort at /home/stbn/git/esp-idf-v4.4.2/components/esp_system/panic.c:402

0x40089649: esp_system_abort at /home/stbn/git/esp-idf-v4.4.2/components/esp_system/esp_system.c:128

0x4008fe81: __assert_func at /home/stbn/git/esp-idf-v4.4.2/components/newlib/assert.c:85

0x4008f1d9: block_merge_prev at /home/stbn/git/esp-idf-v4.4.2/components/heap/heap_tlsf.c:350
 (inlined by) tlsf_free at /home/stbn/git/esp-idf-v4.4.2/components/heap/heap_tlsf.c:874

0x4008faaa: multi_heap_free_impl at /home/stbn/git/esp-idf-v4.4.2/components/heap/multi_heap.c:222

0x40081fa2: heap_caps_free at /home/stbn/git/esp-idf-v4.4.2/components/heap/heap_caps.c:340

0x4008feb1: free at /home/stbn/git/esp-idf-v4.4.2/components/newlib/heap.c:39

0x40168c4f: mesh_free at ??:?

0x40168cf6: esp_mesh_free_context at ??:?

0x40145a73: mesh_recv at ??:?

0x40146e67: esp_mesh_recv_toDS at ??:?

0x400f8593: __mwifi_root_read at /home/stbn/git/esp-mdf/components/mwifi/mwifi.c:1340 (discriminator 8)

0x400e6ed3: tasks::root_upstream_task(void*) at /home/stbn/git/biggrower-fw/build/../main/mesh/tasks.cpp:49

0x4008cdad: vPortTaskWrapper at /home/stbn/git/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131
```

Trace 2:

```
assert failed: insert_free_block heap_tlsf.c:239 (current && "free list cannot have a null entry")

Backtrace:0x40081cca:0x3ffe2d700x4008956d:0x3ffe2d90 0x4008fd65:0x3ffe2db0 0x4008f343:0x3ffe2ed0 0x4008f99a:0x3ffe2ef0 0x40081f3a:0x3ffe2f10 0x4008fd95:0x3ffe2f30 0x40168aff:0x3ffe2f50 0x40168ba6:0x3ffe2f70 0x40145923:0x3ffe2fa0 0x40146d17:0x3ffe2ff0 0x400f82e3:0x3ffe3020 0x400e6e27:0x3ffe30c0 0x4008ccc5:0x3ffe3210

0x40081cca: panic_abort at /home/stbn/git/esp-idf-v4.4.1/components/esp_system/panic.c:402

0x4008956d: esp_system_abort at /home/stbn/git/esp-idf-v4.4.1/components/esp_system/esp_system.c:128

0x4008fd65: __assert_func at /home/stbn/git/esp-idf-v4.4.1/components/newlib/assert.c:85

0x4008f343: insert_free_block at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:239

(inlined by) block_insert at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:269

(inlined by) tlsf_free at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:876

0x4008f99a: multi_heap_free_impl at /home/stbn/git/esp-idf-v4.4.1/components/heap/multi_heap.c:220

0x40081f3a: heap_caps_free at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:340

0x4008fd95: free at /home/stbn/git/esp-idf-v4.4.1/components/newlib/heap.c:39

0x40168aff: mesh_free at ??:?

0x40168ba6: esp_mesh_free_context at ??:?

0x40145923: mesh_recv at ??:?

0x40146d17: esp_mesh_recv_toDS at ??:?

0x400f82e3: __mwifi_root_read at /home/stbn/git/esp-mdf/components/mwifi/mwifi.c:1340 (discriminator 8)
```

Trace 3:

```
assert failed: block_trim_free heap_tlsf.c:377 (block_is_free(block) && "block must be free")

Backtrace:0x40081cca:0x3ffe2e100x4008956d:0x3ffe2e30 0x4008fd65:0x3ffe2e50 0x4008ef0a:0x3ffe2f70 0x4008f935:0x3ffe2f90 0x40081e71:0x3ffe2fb0 0x40081f49:0x3ffe2fd0 0x40082055:0x3ffe3000 0x400f86bd:0x3ffe3020 0x400e6eab:0x3ffe30c0 0x4008ccc5:0x3ffe3210
0x40081cca: panic_abort at /home/stbn/git/esp-idf-v4.4.1/components/esp_system/panic.c:402

0x4008956d: esp_system_abort at /home/stbn/git/esp-idf-v4.4.1/components/esp_system/esp_system.c:128

0x4008fd65: __assert_func at /home/stbn/git/esp-idf-v4.4.1/components/newlib/assert.c:85

0x4008ef0a: block_trim_free at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:377
(inlined by) block_prepare_used at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:460
(inlined by) tlsf_malloc at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:758

0x4008f935: multi_heap_malloc_impl at /home/stbn/git/esp-idf-v4.4.1/components/heap/multi_heap.c:197

0x40081e71: heap_caps_malloc_base at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:147

0x40081f49: heap_caps_realloc_base at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:354

0x40082055: heap_caps_realloc at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:423

0x400f86bd: __mwifi_root_read at /home/stbn/git/esp-mdf/components/mwifi/mwifi.c:1406 (discriminator 6)
```

```
assert failed: block_trim_free heap_tlsf.c:377 (block_is_free(block) && "block must be free")

Backtrace:0x40081cca:0x3ffc8e200x4008956d:0x3ffc8e40 0x4008fd65:0x3ffc8e60 0x4008ef0a:0x3ffc8f80 0x4008f935:0x3ffc8fa0 0x40081e71:0x3ffc8fc0 0x40081e99:0x3ffc8fe0 0x40081ec5:0x3ffc9000 0x4008fd75:0x3ffc9020 0x40085e19:0x3ffc9040 0x401dcd7d:0x3ffc9060 0x401dd4d8:0x3ffc9090 0x401de0bc:0x3ffc90c0 0x401de525:0x3ffc9100 0x401de5db:0x3ffc9140 0x401db043:0x3ffc9180 0x4008ccc5:0x3ffc91b0
0x40081cca: panic_abort at /home/stbn/git/esp-idf-v4.4.1/components/esp_system/panic.c:402

0x4008956d: esp_system_abort at /home/stbn/git/esp-idf-v4.4.1/components/esp_system/esp_system.c:128

0x4008fd65: __assert_func at /home/stbn/git/esp-idf-v4.4.1/components/newlib/assert.c:85

0x4008ef0a: block_trim_free at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:377
(inlined by) block_prepare_used at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:460
(inlined by) tlsf_malloc at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_tlsf.c:758

0x4008f935: multi_heap_malloc_impl at /home/stbn/git/esp-idf-v4.4.1/components/heap/multi_heap.c:197

0x40081e71: heap_caps_malloc_base at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:147

0x40081e99: heap_caps_malloc at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:167

0x40081ec5: heap_caps_malloc_default at /home/stbn/git/esp-idf-v4.4.1/components/heap/heap_caps.c:192

0x4008fd75: malloc at /home/stbn/git/esp-idf-v4.4.1/components/newlib/heap.c:24

0x40085e19: wifi_malloc at /home/stbn/git/esp-idf-v4.4.1/components/esp_wifi/esp32/esp_adapter.c:80

0x401dcd7d: esf_buf_alloc_dynamic at ??:?

0x401dd4d8: esf_buf_alloc at ??:?

0x401de0bc: wDev_IndicateFrame at ??:?

0x401de525: wDev_ProcessRxSucData at ??:?

0x401de5db: wdevProcessRxSucDataAll at ??:?

0x401db043: ppTask at ??:?

0x4008ccc5: vPortTaskWrapper at /home/stbn/git/esp-idf-v4.4.1/components/freertos/port/xtensa/port.c:131
```

### Steps to reproduce.

1. Create a mesh network
2. The root node receives a large number of packets
3. The root node relays those packets using MQTT
4. The root node ends up crashing

Maksons, 2022-09-26:
Have the same bug probably, 
with master branch from :  2022-08-26 13:25 +0800 morris 060f3a3b13241e4c9e160fc61a9995b51c3ba0dc 
when called function:
`aligned_alloc();`

fushun1990, 2022-10-07:
遇到相同问题了
`I (494317) APP_DISPLAY_PIC: free_image_data_timer_cb

assert failed: insert_free_block heap_tlsf.c:239 (current && "free list cannot have a null entry")

Backtrace: 0x4037640a:0x3de2eb90 0x40382455:0x3de2ebb0 0x4038dc01:0x3de2ebd0 0x40389793:0x3de2ecf0 0x40389e02:0x3de2ed10 0x403769a2:0x3de2ed30 0x4038dc31:0x3de2ed50 0x42018276:0x3de2ed70 0x4201b293:0x3de2ed90 0x40386095:0x3de2edb0
0x4037640a: panic_abort at /Users/wangfushun/esp/esp-idf/components/esp_system/panic.c:402

0x40382455: esp_system_abort at /Users/wangfushun/esp/esp-idf/components/esp_system/esp_system.c:128

0x4038dc01: __assert_func at /Users/wangfushun/esp/esp-idf/components/newlib/assert.c:85

0x40389793: insert_free_block at /Users/wangfushun/esp/esp-idf/components/heap/heap_tlsf.c:239
 (inlined by) block_insert at /Users/wangfushun/esp/esp-idf/components/heap/heap_tlsf.c:269
 (inlined by) tlsf_free at /Users/wangfushun/esp/esp-idf/components/heap/heap_tlsf.c:876

0x40389e02: multi_heap_free_impl at /Users/wangfushun/esp/esp-idf/components/heap/multi_heap.c:222

0x403769a2: heap_caps_free at /Users/wangfushun/esp/esp-idf/components/heap/heap_caps.c:361

0x4038dc31: free at /Users/wangfushun/esp/esp-idf/components/newlib/heap.c:39

0x42018276: app_mem_free at /Volumes/work/CLionProjects/menling-display/main/app_mem.c:19

0x4201b293: free_image_data_timer_cb at /Volumes/work/CLionProjects/menling-display/main/app_display/app_display_pic.c:134

0x40386095: vPortTaskWrapper at /Users/wangfushun/esp/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: f9c79b7866bbf0e5

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
`

nateexptr, 2022-12-24:
Hello, can anyone solve this problem, I'm using it with freeRTOS. not sure what going on in this code.

ESP32S3  N8R2
esp-idf 5.0

nateexptr, 2022-12-24:
assert failed: insert_free_block tlsf.c:364 (current && "free list cannot have a null entry")

Backtrace: 0x40375c8e:0x3fce9980 0x40379b25:0x3fce99a0 0x4037f0e1:0x3fce99c0 0x4037e543:0x3fce9ae0 0x4037dec3:0x3fce9b00 0x40375e79:0x3fce9b20 0x40375ea7:0x3fce9b40 0x40375ede:0x3fce9b60 0x4037f0f1:0x3fce9b80 0x4037a37c:0x3fce9ba0 0x4037a630:0x3fce9bc0 0x4037709a:0x3fce9be0 0x403770da:0x3fce9c00 0x4037721d:0x3fce9c30 0x42006909:0x3fce9c50 0x420064a1:0x3fce9c70 0x42006042:0x3fce9c90 0x4200ab75:0x3fce9cb0 0x42012976:0x3fce9cd0 0x42012a02:0x3fce9cf0 0x4200a63f:0x3fce9d20 0x42015f27:0x3fce9d50 0x42011626:0x3fce9d70 0x4200a7e5:0x3fcea080 0x420078e4:0x3fcea0d0 0x4037c4b9:0x3fcea0f0
0x40375c8e: panic_abort at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/esp_system/panic.c:412

0x40379b25: esp_system_abort at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/esp_system/esp_system.c:135

0x4037f0e1: __assert_func at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/newlib/assert.c:78

0x4037e543: insert_free_block at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/tlsf/tlsf.c:364
 (inlined by) block_insert at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/tlsf/tlsf.c:394
 (inlined by) block_trim_free at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/tlsf/tlsf.c:508
 (inlined by) block_prepare_used at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/tlsf/tlsf.c:585
 (inlined by) tlsf_malloc at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/tlsf/tlsf.c:935

0x4037dec3: multi_heap_malloc_impl at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/multi_heap.c:217

0x40375e79: heap_caps_malloc_base at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/heap_caps.c:146

0x40375ea7: heap_caps_malloc at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/heap_caps.c:166

0x40375ede: heap_caps_malloc_default at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/heap/heap_caps.c:191

0x4037f0f1: malloc at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/newlib/heap.c:24

0x4037a37c: xQueueGenericCreate at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/queue.c:454

0x4037a630: xQueueCreateMutex at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/queue.c:571

0x4037709a: lock_init_generic at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/newlib/locks.c:64

0x403770da: lock_acquire_generic at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/newlib/locks.c:118

0x4037721d: _lock_acquire_recursive at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/newlib/locks.c:158

0x42006909: usb_serial_jtag_write at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/vfs/vfs_usb_serial_jtag.c:145

0x420064a1: console_write at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/vfs/vfs_console.c:75

0x42006042: esp_vfs_write at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/vfs/vfs.c:436 (discriminator 4)

0x4200ab75: __swrite at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/stdio.c:94

0x42012976: __sflush_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fflush.c:224

0x42012a02: _fflush_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fflush.c:278

0x4200a63f: __sfvwrite_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251

0x42015f27: __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399

0x42011626: _vfprintf_r at ??:?

0x4200a7e5: printf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/printf.c:56 (discriminator 5)

0x420078e4: myTask at F:/ESP32/ESP32_SDK/frameworks/mycode/hello_world/main/hello_world_main.c:24 (discriminator 1)

0x4037c4b9: vPortTaskWrapper at F:/ESP32/ESP32_SDK/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: fffc40ea4e4e0a98

Rebooting...

adirsingh96, 2023-07-01:
Are  you using MDF_CALLOC to allocate memory? there is some issue in its implementation , using MDF_MALLOC instead of MDF_CALLOC solved my issue .