espressif/esp-mdf #307: mdf 的router例程tcp 线程退出引起问题

yeqinggui, 2022-10-19:
硬件采用c3，测试例程为router，发现wifi 路由器重启后，根节点无法再次进入事件MDF_EVENT_MWIFI_ROOT_GOT_IP，parentmac打印出来都是0，日志如下：
W (29830) wifi:m f probe req l=0

I (29837) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (29842) wifi:state: init -> auth (b0)
[0;32mI (29846) [router_example, 2556]: Parent is disconnected on station interface[0m
h屙螰V?c弰)Hh屙R>FFBV#?H[0;32mI (16138) [mwifi, 188]: Parent is disconnected, reason: 205[0m
I (16138) mesh: [wifi]disconnected reason:205(), continuous:4/max:12, root, vote(,stopped)<><>
[0;32mI (16139) [router_example, 2521]: event_loop_cb, event: 8[0m
[0;32mI (16148) [router_example, 2556]: Parent is disconnected on station interface[0m
I (16191) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (16192) wifi:state: init -> auth (b0)
I (16791) wifi:state: auth -> assoc (0)
I (17348) wifi:state: assoc -> run (10)
I (18441) wifi:connected with sk-wifi, aid = 1, channel 1, BW20, bssid = a4:39:b3:c8:52:f2
I (18442) wifi:security: WPA2-PSK, phy: bgn, rssi: -48
I (18444) wifi:pm start, type: 0

I (18447) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (18456) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (18463) mesh: <nvs>write layer:1
[0;32mI (18464) [mwifi, 175]: Parent is connected[0m
[0;32mI (18469) [mwifi, 258]: State represents: 0[0m
[0;32mI (18474) [router_example, 2521]: event_loop_cb, event: 7[0m
[0;32mI (18480) [router_example, 2530]: Parent is connected on station interface[0m
[0;32mI (18487) [router_example, 2537]: 

root 
[0m
I (18475) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
[0;32mI (18494) [router_example, 2521]: event_loop_cb, event: 11[0m
[0;32mI (18505) [router_example, 2521]: event_loop_cb, event: 14[0m
[0;32mI (18512) [router_example, 2634]: root address
[0m
I (18517) wifi:BcnInt:102400, DTIM:1
I (18521) mesh: <flush>root
I (18523) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)

yeqinggui, 2022-10-19:
[image]

yeqinggui, 2022-10-19:
[image]

Jiangyafeng, 2022-10-20:
@yeqinggui  请在 MDF_EVENT_MWIFI_PARENT_CONNECTED 时间里面添加如下操作：
`        case MDF_EVENT_MWIFI_PARENT_CONNECTED:
            MDF_LOGI("Parent is connected on station interface");

            if (esp_mesh_is_root()) {
                esp_netif_dhcpc_stop(netif_sta);
                esp_netif_dhcpc_start(netif_sta);
            }

            break;
`

yeqinggui, 2022-10-21:
> 

谢谢，原先case MDF_EVENT_MWIFI_PARENT_CONNECTED: 已经有 esp_netif_dhcpc_start(netif_sta);，但没有esp_netif_dhcpc_stop(netif_sta);，可能这个原因造成二次连接无法get ip，

yeqinggui, 2022-10-21:
[image]
试了一下，依然不行

Jiangyafeng, 2022-10-26:
建议您可以直接在 router  demo 中加上上面的代码，做一下测试， 
link：https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router