espressif/esp-mdf #310: esp32-C3系列可否改wifi的天线功率

yeqinggui, 2022-10-27:
目前测试mesh，父节点wifi 的ssi 距离5米不到就只有-65左右，中间没任何东西阻挡，若是再隔着墙就通信不稳定，用的是内置板上天线，这个天线发射功率是否可以程序控制？

FPOOOOOOO, 2022-10-27:
好像10-20dB是idf.py预编译的，如果要直接程序变量动态的话，你要找一下宏在哪里。我感觉需要重启才行。

yeqinggui, 2022-10-28:
10-20dB 理论上应该是够，但实际测试，5米就衰减到60，是不是这种模块做ap，wifi信号都不强

FPOOOOOOO, 2022-10-28:
我用的是Wroom-32D，自己外接天线，5米-37是正常的。5米-60的话，确实是有些低了。板载IPEX也没有这么低

yeqinggui, 2022-10-31:
> 我用的是Wroom-32D，自己外接天线，5米-37是正常的。5米-60的话，确实是有些低了。板载IPEX也没有这么低

你好，有测试板载做父节点，5米一般达到多少呢？

FPOOOOOOO, 2022-10-31:
-37~-44

yeqinggui, 2022-11-01:
> -37~-44

奇怪我手上模组esp32-c3-wroom-02，基本都是-60db左右，不知道是不是发射功率设置小了

FPOOOOOOO, 2022-11-01:
你看看idf.py menuconfig里发射功率是不是20，或者就换成esp32-WROOM-32D，哈哈

yeqinggui, 2022-11-02:
非常感谢指导，在 menuconfig里边还没找到对应得功率大小是设置，还在查

FPOOOOOOO, 2022-11-02:
我是MAC，在终端里直接输入idf.py menuconfig，会出现micropython的界面，然后在PHY路径下就有
[image]

yeqinggui, 2022-11-02:
谢谢，找到了，改为10db试试
[image]

yeqinggui, 2022-11-03:
实际测试，改这个没舍区别，信号依然差

closed 2022-11-03 by yeqinggui