espressif/esp-mdf #315: mdf router 发送报内存不足

yeqinggui, 2022-11-03:
子节点每隔20秒发送一包给父接点，父节点再发回应答包，一段时间后，发现日志报ESP_ERR_MESH_NO_MEMORY，是不是底层mesh发送不成功，没扔掉缓存数据？
D (73354263) [router_example, 2294]: Node receive: 58:cf:79:21:b1:40, size: 102[0m
[0;32mI (73354264) [router_example, 1516]: receive cmd= ping
[0m
D (73367544) [router_example, 2294]: Node receive: a0:76:4e:4d:d8:fc, size: 54[0m
[0;32mI (73367545) [router_example, 1373]: receive son node ping:a0:76:4e:4d:d8:fc[0m
[0;32mI (73367700) [router_example, 1189]: send ping to son, size: 102, data: {"header":{"wifimac":"a0:76:4e:4d:d8:fc","cmd":"ping"},"body":{"timestamp":1667449574,"fdeviceno":12}}[0m
I (73368952) mesh: 5116<assoc>parent layer:1, channel:1, rssi:-50, assoc:1, rssi threshold<-60,-65,-70>
[0;33mW (73380104) [mwifi, 705]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, dest_addr: 58:cf:79:21:b1:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfcb8, size: 54[0m
[0;33mW (73380111) [mwifi, 958]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, data_flag: 0x22, dest_mac: 58:cf:79:21:b1:40[0m
[0;32mI (73380123) [router_example, 1218]: <ESP_ERR_MESH_TIMEOUT> mwifi_write ERR!
[0m
[0;32mI (73382102) [router_example, 3669]: mymac: 68:67:25:56:56:48[0m
[0;32mI (73382102) [router_example, 3678]:  wifi_sta_list  1
[0m
[0;32mI (73382103) [router_example, 3683]: son mac: a0:76:4e:4d:d8:fc[0m
[0;32mI (73382110) [router_example, 3692]: root status:1, channel: 1,bkchannel: 73381633 ,mylayer: 2,p-no: 11, p-rssi: -51,sonnum: 1,free heap: 127784[0m
[0;32mI (73382123) [router_example, 3697]: mwifi_is_connected,cmd=0[0m
D (73388411) [router_example, 2294]: Node receive: a0:76:4e:4d:d8:fc, size: 54[0m
[0;32mI (73388411) [router_example, 1373]: receive son node ping:a0:76:4e:4d:d8:fc[0m
[0;33mW (73393416) [mwifi, 705]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, dest_addr: a0:76:4e:4d:d8:fc, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfd10, size: 102[0m
[0;33mW (73393423) [mwifi, 958]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, data_flag: 0x22, dest_mac: a0:76:4e:4d:d8:fc[0m
[0;32mI (73393435) [router_example, 1185]: <ESP_ERR_MESH_TIMEOUT> mwifi_write ERR!
[0m
[0;33mW (73401103) [mwifi, 705]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, dest_addr: 58:cf:79:21:b1:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcc0c28, size: 54[0m
[0;33mW (73401110) [mwifi, 958]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, data_flag: 0x22, dest_mac: 58:cf:79:21:b1:40[0m
[0;32mI (73401122) [router_example, 1218]: <ESP_ERR_MESH_TIMEOUT> mwifi_write ERR!
[0m
D (73409711) [router_example, 2294]: Node receive: a0:76:4e:4d:d8:fc, size: 54[0m
[0;32mI (73409712) [router_example, 1373]: receive son node ping:a0:76:4e:4d:d8:fc[0m
[0;33mW (73414717) [mwifi, 705]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, dest_addr: a0:76:4e:4d:d8:fc, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfd10, size: 102[0m
[0;33mW (73414724) [mwifi, 958]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, data_flag: 0x22, dest_mac: a0:76:4e:4d:d8:fc[0m
[0;32mI (73414736) [router_example, 1185]: <ESP_ERR_MESH_TIMEOUT> mwifi_write ERR!
[0m
[0;33mW (73422103) [mwifi, 705]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, dest_addr: 58:cf:79:21:b1:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfcb8, size: 54[0m
[0;33mW (73422110) [mwifi, 958]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, data_flag: 0x22, dest_mac: 58:cf:79:21:b1:40[0m
[0;32mI (73422122) [router_example, 1218]: <ESP_ERR_MESH_TIMEOUT> mwifi_write ERR!
[0m
[0;32mI (73442102) [router_example, 3669]: mymac: 68:67:25:56:56:48[0m
[0;32mI (73442102) [router_example, 3678]:  wifi_sta_list  1
[0m
[0;32mI (73442103) [router_example, 3683]: son mac: a0:76:4e:4d:d8:fc[0m
[0;32mI (73442110) [router_example, 3692]: root status:1, channel: 1,bkchannel: 73441537 ,mylayer: 2,p-no: 11, p-rssi: -52,sonnum: 1,free heap: 121756[0m
[0;32mI (73442123) [router_example, 3697]: mwifi_is_connected,cmd=0[0m
[0;33mW (73443103) [mwifi, 705]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, dest_addr: 58:cf:79:21:b1:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfcb8, size: 54[0m
[0;33mW (73443110) [mwifi, 958]: <ESP_ERR_MESH_TIMEOUT> Node failed to send packets, data_flag: 0x22, dest_mac: 58:cf:79:21:b1:40[0m
[0;32mI (73443122) [router_example, 1218]: <ESP_ERR_MESH_TIMEOUT> mwifi_write ERR!
[0m
[0;33mW (73459104) [mwifi, 695]: <ESP_ERR_MESH_NO_MEMORY> esp_mesh_send[0m
[0;33mW (73459204) [mwifi, 695]: <ESP_ERR_MESH_NO_MEMORY> esp_mesh_send[0m
[0;33mW (73459305) [mwifi, 695]: <ESP_ERR_MESH_NO_MEMORY> esp_mesh_send[0m
[0;33mW (73459406) [mwifi, 705]: <ESP_ERR_MESH_NO_MEMORY> Node failed to send packets, dest_addr: 58:cf:79:21:b1:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfcb8, size: 54[0m
[0;33mW (73459413) [mwifi, 958]: <ESP_ERR_MESH_NO_MEMORY> Node failed to send packets, data_flag: 0x22, dest_mac: 58:cf:79:21:b1:40[0m
[0;32mI (73459425) [router_example, 1218]: <ESP_ERR_MESH_NO_MEMORY> mwifi_write ERR!
[0m
[0;33mW (73480104) [mwifi, 695]: <ESP_ERR_MESH_NO_MEMORY> esp_mesh_send[0m
[0;33mW (73480204) [mwifi, 695]: <ESP_ERR_MESH_NO_MEMORY> esp_mesh_send[0m
[0;33mW (73480305) [mwifi, 695]: <ESP_ERR_MESH_NO_MEMORY> esp_mesh_send[0m
[0;33mW (73480406) [mwifi, 705]: <ESP_ERR_MESH_NO_MEMORY> Node failed to send packets, dest_addr: 58:cf:79:21:b1:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3fcbfcb8, size: 54[0m
[0;33mW (73480413) [mwifi, 958]: <ESP_ERR_MESH_NO_MEMORY> Node failed to send packets, data_flag: 0x22, dest_mac: 58:cf:79:21:b1:40[0m
[0;32mI (73480425) [router_example, 1218]: <ESP_ERR_MESH_NO_MEMORY> mwifi_write ERR!

yeqinggui, 2022-11-04:
这边有乐鑫技术支持吗？有问题如何对接呢？这个多次发送时底层mesh超时自发，如何改底层这部分呢？

tomyqg, 2022-11-06:
放弃吧

Jiangyafeng, 2022-11-07:
@yeqinggui  这个是因为节点前面一直发送不出去数据，可以看到 <ESP_ERR_MESH_TIMEOUT> Node failed to send packets，这个log。然后一直还在不停的发送，从而导致底层 mesh buffer满了，从而后面才打印了 <ESP_ERR_MESH_NO_MEMORY> 。所以需要先解决为什么前面发送不成功原因。

yeqinggui, 2022-11-07:
发送不出数据应该是通信连接问题吧，那么这种通信异常，底层为何不扔掉数据？

Jiangyafeng, 2022-11-08:
是因为你的应用层一直调用 esp_mesh_send 发送，节点又发送不出去从而导致的。这个时候你应该做一个判断，从而停止应用层的发送，等待消息不阻塞，能够通讯成功之后，再去发送数据。

yeqinggui, 2022-11-09:
> 是因为你的应用层一直调用 esp_mesh_send 发送，节点又发送不出去从而导致的。这个时候你应该做一个判断，从而停止应用层的发送，等待消息不阻塞，能够通讯成功之后，再去发送数据。

有做mwifi_is_connected()判断，ture 才发送，那么若是这个判断无效，要通过写mwifi_write写后返回才知道异常，那如何避免一直写？最终还不如底层没做这个限制？

Leadrive, 2024-03-21:
今天也碰到了，就是前面出现10几次ESP_ERR_MESH_TIMEOUT后，就会出现ESP_ERR_MESH_NO_MEMORY，此时并没收到ESP_MESH_CHILD_DISCONNECTED消息，链接并没断开，只是发不出去

tonyshuang, 2024-03-21:
您好，信件已收到。