espressif/esp-mdf #320: Win10环境中，先安装了IDF，后安装了MDF，但是MDF编译不能通过，通过vscode的插件可以编译并下载成功

Convict0830, 2023-01-04:
components/esp_http_server/include -ID:/ESP32_IDF/Espressif/frameworks/esp-idf-v4.1.4/components/app_update/include -ID:/ESP32_IDF/Espressif/frameworks/esp-idf-v4.1.4/components/bootloader_support/include -ID:/ESP32_IDF/Esp32MDF/esp-mdf/components/mconfig/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.1.4\" -DMDF_VER=\"v1.0-158-g8d0b79a\" -DESP_PLATFORM -MD -MT esp-idf/mlink/CMakeFiles/__idf_mlink.dir/mlink_handle.c.obj -MF esp-idf\mlink\CMakeFiles\__idf_mlink.dir\mlink_handle.c.obj.d -o esp-idf/mlink/CMakeFiles/__idf_mlink.dir/mlink_handle.c.obj   -c D:/ESP32_IDF/Esp32MDF/esp-mdf/components/mlink/mlink_handle.c
In file included from D:/ESP32_IDF/Esp32MDF/esp-mdf/components/mlink/mlink_handle.c:16:
D:/ESP32_IDF/Esp32MDF/esp-mdf/components/mwifi/include/mwifi.h:265:59: error: unknown type name 'mesh_chain_layer_t'; did you mean 'esp_chip_model_t'?
 esp_err_t esp_wifi_vnd_mesh_get(mesh_assoc_t *mesh_assoc, mesh_chain_layer_t *mesh_chain);
                                                           ^~~~~~~~~~~~~~~~~~
                                                           esp_chip_model_t
D:/ESP32_IDF/Esp32MDF/esp-mdf/components/mlink/mlink_handle.c: In function 'mlink_handle_get_config':
D:/ESP32_IDF/Esp32MDF/esp-mdf/components/mlink/mlink_handle.c:627:5: error: unknown type name 'mesh_chain_layer_t'; did you mean 'esp_chip_model_t'?
     mesh_chain_layer_t mesh_chain;
     ^~~~~~~~~~~~~~~~~~
     esp_chip_model_t
D:/ESP32_IDF/Esp32MDF/esp-mdf/components/mlink/mlink_handle.c:630:5: error: implicit declaration of function 'esp_wifi_vnd_mesh_get'; did you mean 'esp_wifi_get_ant'? [-Werror=implicit-function-declaration]
     esp_wifi_vnd_mesh_get(&mesh_assoc, &mesh_chain);
     ^~~~~~~~~~~~~~~~~~~~~
     esp_wifi_get_ant
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

FPOOOOOOO, 2023-01-11:
我的理解是通过VSCODE安装套件的话，IDF能够成功，MDF放在和IDF同级的目录里就可以编译了，我这里是这样的

Convict0830, 2023-01-11:
> 我的理解是通过VSCODE安装套件的话，IDF能够足够成功，MDF放在那里和IDF同级的目录里就可以编译了，我这里是这样的

嗯，就很奇怪，有时候新建的工程就一直在re-running cmake 无法编译，有时候突然又可以编译了

sxc1143106070, 2024-04-22:
> 我的理解是通过VSCODE安装套件的话，IDF能够成功，MDF放在和IDF同级的目录里就可以编译了，我这里是这样的

你好，我目前 VScode + IDF 可以正常使用，现在向添加MDF到VScode，能麻烦你具体说一下怎么做吗

tonyshuang, 2024-04-22:
您好，信件已收到。

FPOOOOOOO, 2024-04-22:
> > 我的理解是通过VSCODE安装套件的话，IDF能够成功，MDF放在和IDF同级的目录里就可以编译了，我这里是这样的
> 
> 你好，我目前 VScode + IDF 可以正常使用，现在向添加MDF到VScode，能麻烦你具体说一下怎么做吗

乐鑫官网有MDF的操作流程可以参考，可以试试直接git clone mdf的仓库到idf的同级目录下，这样idf.py build都能使用

iwonako, 2024-11-21:
> 乐鑫官网有MDF的操作流程可以参考，可以试试直接git clone mdf的仓库到idf的同级目录下，这样idf.py build都能使用

我是通过 idf 插件里面的 `install esp-mdf` 安装的，但它安装的路径是在`\Espressif\`下，同级目录是这个吗`\Espressif\frameworks\v5.3.1`。

安装后打开例程后点 `SDK configure editor` ，报错 `The component mdns could not be found`，点编译也失败了