espressif/esp-mdf #328: mconfig_queue_read超时问题，修改队列长度为5能规避。

Fuhua-Chen, 2023-03-26:
**1. 问题描述**
    2年前的项目，持续到22年，均没有出现问题（期间也有更换SDK和电脑），23年使用新电脑最新release v4.4.3编译，出现mconfig_queue_read 超时问题，出现概率几乎100%，偶尔能成功接收到mconfig master的信息。

**2. 问题排查**
    2.1. 因为应用层代码没有变动，而电脑和SDK均更新，因此首先排查mconfig_queue.c。
    2.2. 发现超时日志代码前，仅仅创建了一个g_mconfig_queue进行接收。而创建过程没有发生错误。
    2.3. 发现创建队列仅使用了1个数据长度作为队列长度， 猜想如果接收数据过程异常、数据量大、主机发送数据频繁，可能导致数据丢失问题。
    2.4. 修改队列长度为5，多次测试后发现虽然超时仍然发生，但可以在重试mconfig_queue_read过程中正确接收一次。

**3. 问题解决**
   **请教此问题应如何调试确认问题点。**

Jiangyafeng, 2023-03-27:
@Fuhua-Chen  请问一下，是否可以在 v4.3.1  上还会出现这样的问题吗？

还有请提供一下设备端 log 信息，还有是否可直接在 mconfig_example 下就能复现你的问题那？

Fuhua-Chen, 2023-10-21:
目前SDK因为arduio-esp32的要求，变更SDK到v4.4.3了，也发现这个版本的SDK把队列长度改为5了，但是这个接收不到消息的问题更加严重了，几乎没办法配网