espressif/esp-mdf #332: win10 VScode中idf编译报错 ninja: build stopped: subcommand failed.

Noviceontheway, 2023-11-16:
VScode中idf版本：4.4.6（通过vscode插件安装）
mdf也是通过vscode安装的为master版本

在编译到aliyun_sdk的时候报错ninja: build stopped: subcommand failed.

[1091/1120] Building C object esp-idf/aliyun_sdk/CMakeFiles/__idf_aliyun_sdk.dir/coap/aliyun_coap.c.obj
FAILED: esp-idf/aliyun_sdk/CMakeFiles/__idf_aliyun_sdk.dir/coap/aliyun_coap.c.obj 
ccache C:\vscode-esp\Tools\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DWITH_POSIX -IC:/Users/work/Desktop/get-started/build/config -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/coap/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/message/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/mqtt/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/ota/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/platform/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/sign/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/platform/idf -IC:/vscode-esp/esp-idf/components/newlib/platform_include -IC:/vscode-esp/esp-idf/components/freertos/include -IC:/vscode-esp/esp-idf/components/freertos/include/esp_additions/freertos -IC:/vscode-esp/esp-idf/components/freertos/port/xtensa/include -IC:/vscode-esp/esp-idf/components/freertos/include/esp_additions -IC:/vscode-esp/esp-idf/components/esp_hw_support/include -IC:/vscode-esp/esp-idf/components/esp_hw_support/include/soc -IC:/vscode-esp/esp-idf/components/esp_hw_support/include/soc/esp32s2 -IC:/vscode-esp/esp-idf/components/esp_hw_support/port/esp32s2/. -IC:/vscode-esp/esp-idf/components/esp_hw_support/port/esp32s2/private_include -IC:/vscode-esp/esp-idf/components/heap/include -IC:/vscode-esp/esp-idf/components/log/include -IC:/vscode-esp/esp-idf/components/lwip/include/apps -IC:/vscode-esp/esp-idf/components/lwip/include/apps/sntp -IC:/vscode-esp/esp-idf/components/lwip/lwip/src/include -IC:/vscode-esp/esp-idf/components/lwip/port/esp32/include -IC:/vscode-esp/esp-idf/components/lwip/port/esp32/include/arch 
-IC:/vscode-esp/esp-idf/components/soc/include -IC:/vscode-esp/esp-idf/components/soc/esp32s2/. -IC:/vscode-esp/esp-idf/components/soc/esp32s2/include -IC:/vscode-esp/esp-idf/components/hal/esp32s2/include -IC:/vscode-esp/esp-idf/components/hal/include -IC:/vscode-esp/esp-idf/components/hal/platform_port/include -IC:/vscode-esp/esp-idf/components/esp_rom/include -IC:/vscode-esp/esp-idf/components/esp_rom/include/esp32s2 -IC:/vscode-esp/esp-idf/components/esp_rom/esp32s2 -IC:/vscode-esp/esp-idf/components/esp_common/include -IC:/vscode-esp/esp-idf/components/esp_system/include -IC:/vscode-esp/esp-idf/components/esp_system/port/soc -IC:/vscode-esp/esp-idf/components/esp_system/port/public_compat -IC:/vscode-esp/esp-idf/components/xtensa/include -IC:/vscode-esp/esp-idf/components/xtensa/esp32s2/include -IC:/vscode-esp/esp-idf/components/driver/include -IC:/vscode-esp/esp-idf/components/driver/esp32s2/include -IC:/vscode-esp/esp-idf/components/esp_pm/include -IC:/vscode-esp/esp-idf/components/esp_ringbuf/include -IC:/vscode-esp/esp-idf/components/efuse/include -IC:/vscode-esp/esp-idf/components/efuse/esp32s2/include -IC:/vscode-esp/esp-idf/components/vfs/include -IC:/vscode-esp/esp-idf/components/esp_wifi/include -IC:/vscode-esp/esp-idf/components/esp_event/include -IC:/vscode-esp/esp-idf/components/esp_netif/include -IC:/vscode-esp/esp-idf/components/esp_eth/include -IC:/vscode-esp/esp-idf/components/tcpip_adapter/include -IC:/vscode-esp/esp-idf/components/esp_phy/include -IC:/vscode-esp/esp-idf/components/esp_phy/esp32s2/include -IC:/vscode-esp/esp-idf/components/esp_ipc/include -IC:/vscode-esp/esp-idf/components/app_trace/include -IC:/vscode-esp/esp-idf/components/esp_timer/include -IC:/vscode-esp/esp-idf/components/mbedtls/port/include -IC:/vscode-esp/esp-idf/components/mbedtls/mbedtls/include -IC:/vscode-esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/vscode-esp/esp-idf/components/app_update/include -IC:/vscode-esp/esp-idf/components/spi_flash/include -IC:/vscode-esp/esp-idf/components/bootloader_support/include -IC:/vscode-esp/esp-idf/components/nvs_flash/include -IC:/vscode-esp/esp-idf/components/esp_http_client/include -IC:/vscode-esp/esp-idf/components/nghttp/port/include -IC:/vscode-esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/vscode-esp/esp-idf/components/json/cJSON -IC:/vscode-esp/esp-idf/components/mqtt/esp-mqtt/include -IC:/vscode-esp/esp-idf/components/tcp_transport/include -IC:/vscode-esp/esp-idf/components/esp-tls -IC:/vscode-esp/esp-idf/components/esp-tls/esp-tls-crypto -IC:/vscode-esp/esp-idf/components/coap/port/include -IC:/vscode-esp/esp-idf/components/coap/libcoap/include -IC:/vscode-esp/Tools/esp-mdf/components/mcommon/include -IC:/vscode-esp/Tools/esp-mdf/components/mwifi/include -IC:/vscode-esp/Tools/esp-mdf/components/third_party/miniz -IC:/vscode-esp/Tools/esp-mdf/components/mlink/include -IC:/vscode-esp/Tools/esp-mdf/components/mespnow/include -IC:/vscode-esp/Tools/esp-mdf/components/mupgrade/include -IC:/vscode-esp/esp-idf/components/mdns/include -IC:/vscode-esp/esp-idf/components/console -IC:/vscode-esp/esp-idf/components/esp_http_server/include -IC:/vscode-esp/Tools/esp-mdf/components/mconfig/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=C:/Users/work/Desktop/get-started=. -fmacro-prefix-map=C:/vscode-esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.6\" -DMDF_VER=\"v1.0-160-g354d0bf\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/aliyun_sdk/CMakeFiles/__idf_aliyun_sdk.dir/coap/aliyun_coap.c.obj -MF esp-idf\aliyun_sdk\CMakeFiles\__idf_aliyun_sdk.dir\coap\aliyun_coap.c.obj.d -o 
esp-idf/aliyun_sdk/CMakeFiles/__idf_aliyun_sdk.dir/coap/aliyun_coap.c.obj -c C:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/coap/aliyun_coap.c
C:/vscode-esp/Tools/esp-mdf/components/third_party/aliyun_sdk/coap/aliyun_coap.c:16:10: fatal error: coap.h: No such file or directory  
 #include "coap.h"
          ^~~~~~~~
compilation terminated.
[1096/1120] Building C object esp-idf/aliyun_sdk/CMakeFiles/__idf_aliyun_sdk.dir/aliyun_kv.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

哪位老哥能帮我看一下问题

Cyfarw9dd, 2024-01-15:
同问，今天也出现了这个问题，直接使用例程进行编译的，不知道是否需要在编译前插上板子配置好端口

tonyshuang, 2024-01-15:
您好，信件已收到。

iwonako, 2024-11-21:
我配置 SDK congifure editor 就报错了，有遇到这个问题吗 @Noviceontheway

iwonako, 2024-11-21:
> 同问，今天也出现了这个问题，直接使用例程进行编译的，不知道是否需要在编译前插上板子配置好端口

问下是怎么安装的？我是通过 idf 插件里面的 `install esp-mdf` 安装的，安装后打开例程点 `SDK configure editor` ，报错 `The component mdns could not be found`，点编译也失败了

Cyfarw9dd, 2024-11-21:
> > 同问，今天也出现了这个问题，直接使用例程进行编译的，不知道是否需要在编译前插上板子配置好端口
> 
> 问下是怎么安装的？我是通过 idf 插件里面的 `install esp-mdf` 安装的，安装后打开例程点 `SDK configure editor` ，报错 `The component mdns could not be found`，点编译也失败了

esp-idf 在5.0以后的版本不再支持mdns组件，你需要自己在组件管理器中启用这个组件，下一次开始编译的时候mdns会自动下载，可以查看这个链接添加mdns组件 https://docs.espressif.com/projects/esp-idf/zh_CN/v5.3.1/esp32/api-reference/protocols/mdns.html

iwonako, 2024-11-21:
> esp-idf 在5.0以后的版本不再支持mdns组件，你需要自己在组件管理器中启用这个组件，下一次开始编译的时候mdns会自动下载，可以查看这个链接添加mdns组件 https://docs.espressif.com/projects/esp-idf/zh_CN/v5.3.1/esp32/api-reference/protocols/mdns.html

试了，编译报错`Failed to resolve component 'mdns'`，打开 SDK 也是报错。

Cyfarw9dd, 2024-11-21:
那我就不是很清楚了，可以问下gpt看看