espressif/esp-mdf #338: 这个情况怎么解决，装了好多次了

lszpsf0, 2024-11-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

See also "E:/esp/esp-mdf/examples/get-started/build/CMakeFiles/CMakeOutput.log".
HINT: The component mdns could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the E:\esp\esp-mdf\examples\get-started\build\log\idf_py_stderr_output_7040 and E:\esp\esp-mdf\examples\get-started\build\log\idf_py_stdout_output_7040

### Actual behavior (suspected bug)

[]([image]

### Error logs or terminal output

```shell
。
```

### Steps to reproduce the behavior

。

### Project release version

。

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win10

### Shell

CMD

### Additional context

。

tonyshuang, 2024-11-14:
您好，信件已收到。

iwonako, 2024-11-21:
same issue